motioneye-project/motioneye #141: motioneye service fails to start

cferrall, 2016-03-04:
I installed motioneye on a RaspberryPi and everything was great.  Last night the service stopped and now the service will not start.  It hangs on starting motioneye for about a minute and restarts.

Any suggestions?

ccrisan, 2016-03-04:
What OS version do you use? If you have SSH for a minute or so, please log in and do a `cat /var/log/motioneye.log` to see what the problem is. My bet is that you have removed some media files and the `/data/output` directory got removed as well (a bug that I recently fixed).

cferrall, 2016-03-04:
Thanks for the quick reply.   I'll check the version but it was up to date
Wednesday night.  I am able to mount the SD card if connected to another
machine.  What directory would be missing?  I did enable motion recording
the night this occurred.  One of the last things I did was delete the
recordings.

Chris

ccrisan, 2016-03-04:
The /data is the local mount point. Check for the `/output` folder in the root of the third partition's filesystem. It's probably gone anyway. Just create it and you'll be able to boot the OS again.

cferrall, 2016-03-04:
Sweet.

Thanks

ccrisan, 2016-03-05:
I'll close the issue now, as the root cause has been identified and fixed.

closed 2016-03-05 by ccrisan