motioneye-project/motioneye #142: Using Motioneye for a while, what I miss

josejnet, 2016-03-05:
Hey there
I've using motioneye for a while. It's very stable. Even running on a Rp1 over wifi it's incredible stable. Thank you so much Carlin, I really appreciate your work. 

I use it to control in&outs in a small building at strange hours. It works fine. As it's very stable if no one is going in or going out the system does nothing, wich is great, but, ¿If the system suddenly cracks? How do I know the system is up or down?

What I miss: As I set a working schedule to avoid notifications during normal hours, I think it'll be a good idea to have a "Starting shift message with a picture" at the beginning and a "Shift finished message with a picture".
This way, we could check if:
a) System is working or not.
b) Camera is set in the right position.

Now the camera is reachable very easy, but I plan to hide, so "what I miss" would be even more interesting.

I hope you like my idea, have a good day.

ccrisan, 2016-03-05:
Hi,

I understand your concerns and the reasoning behind your request. Unfortunately interfacing with the motion daemon is not as easy as it seems. Everything related to the working schedule is fabricated outside of motion and motion detection control is enabled/disabled by motionEye. I'll think of a way to achieve what you need at the time I'll be developing the manual snapshot functionality.

josejnet, 2016-03-05:
Thanks for your reply. I understand it is not easy. 

The manual snapshot looks interesting. You could also try to develope an "auto snapshot" with several timers/alarms. That would not interfere with the motion daemon, it's just a scheduled snapshot (including notifications and uploads).

Thanks!

/J

sambul13, 2016-04-16:
I just installed MotionEye, and there are some issues:

a) FAQ doesn't say how to access it 1st time. It should say: 
Open localhost:8765 in Firefox, default login: admin / no password

b) email notifications for motion detection events (as Motion is for web surveillance, not a webcam stream watcher package).
If its hard to communicate with Motion daemon, may be its possible to update motion.cfg file, when a user makes changes through MotionEye GUI, and then restart Motion daemon?

c) H264 RTSP IP Cam support. I thought it was the purpose to compile Motion from Mr-Dave's repo, but this key feature is missing from webUI. When adding RTSP cam, it says _RTSP not supported_. Here's some config [example](http://www.jonathanlind.se/14-artiklar/90-motion-with-rtsp-camera) for RTSP Cam. Another thread has some Mr.Dave [advice](http://askubuntu.com/questions/514828/how-can-i-access-the-h264-stream-from-my-ip-camera-with-motion) on setting and recording H264 RTSP webcam for users.