motioneye-project/motioneye #143: Webhook Lag

legotheboss, 2016-03-08:
Hi,

Thanks for the great software. After updating from 0.29 to 0.31.3, I have been experiencing very laggy webhook requests. I have HTTP GET configured via IFTTT and everytime motion is detected, it would previously send the notification instantly. Now, after all the motion has settled (the camera goes from red to normal in the web view), motioneye sends the notification. How can I get the notification to be sent as soon as the motion is detected, like before versions of the software...

ccrisan, 2016-03-08:
The notification should be sent as soon as motion is detected. What hardware are you on?

legotheboss, 2016-03-08:
Raspberry Pi B+

ccrisan, 2016-03-08:
The only possible explanation to your problem is not that the webhook is delayed, but rather the motion detection status indication is now quicker. Please run some tests by enabling the `Show Frame Changes` option and setting `Motion Gap`, `Captured Before`, `Captured After`, and `Minimum Motion Frames` to 1.

legotheboss, 2016-03-09:
@ccrisan I changed the variables you told me to and now the notification works as it is supposed to. Thx for the help :)

closed 2016-03-25 by legotheboss