motioneye-project/motioneye #144: Multiple working_schedule range

MrPumo, 2016-03-08:
Please consider adding multiple start-stop time range for each day.
Thx

heffronc, 2016-03-27:
>Please consider adding multiple start-stop time range for each day.
Thx

I second this request! That would be an awesome feature.

traiger, 2017-03-21:
Third. Thanks for considering.

nicholas1230, 2017-08-26:
Would be a great feature to have!

solarplex, 2018-01-24:
Would very much appreciate this feature

oliveconstant, 2021-04-11:
As a workaround, you could pass the event time to the command:

```bash
/usr/local/bin/motion_detected.sh %H:%M
```

And then handle the scheduling logic inside `motion_detected.sh`, e.g.:

```bash
# "Do not disturb" time ranges, endpoints *excluded*
DND=(
  11:30-14:00
  17:30-23:30
)

# Check do not disturb ranges
for timespan in "${DND[@]}"; do
  start="${timespan%-*}"
  end="${timespan#*-}"
  if [[ "${1}" > "${start}" && "${1}" < "${end}" ]]; then
    exit 0
  fi
done

# Executed only if not in any DND range
echo "Motion detected!"
```