motioneye-project/motioneye #145: Re-authorizing Google Drive

hansen-m, 2016-03-09:
Uploading to Google Drive was working great for me but after adding a new camera and changing some additional settings it seemed to stop working, although clicking 'Test Service' stated that the connection was successful. So I removed the app from Google Drive but I'm not able to re-authorize it, MotionEye just continues to state that the token has been revoked, even though it is a new auth token.

Is there a trick to re-authorize it for Google Drive? Or is it not possible?

hansen-m, 2016-03-09:
Actually it seems that no matter what I enter as a key it claims the token has been revoked.

legotheboss, 2016-03-09:
Have you tried completely removing and reinstalling motioneye?

hansen-m, 2016-03-09:
No, is that necessary?

legotheboss, 2016-03-09:
I would, just in case... It might clear some buggy config files

hansen-m, 2016-03-09:
Seems like a shame when everything else is setup and working perfectly. Could I remove or edit the config instead?

Wondering if this is even possible to re-authorize or if it is limited by Google Drive based on the Client ID?

https://github.com/ccrisan/motioneye/blob/master/motioneye/uploadservices.py#L131

hansen-m, 2016-03-10:
Removing `/data/etc/uploadservices.json` and re-configuring the upload settings seemed to fix the issue.

closed 2016-03-10 by hansen-m