motioneye-project/motioneye #148: System time

Winki43, 2016-03-13:
Hi
How can I change system time? My System has a timestamp from 1970.

ccrisan, 2016-03-14:
I'm assuming this is about motionEyeMake sure your PI has access to the internet and that it can resolve domain names through a DNS server.

Winki43, 2016-03-15:
Hi and thank you for this quick response.
Yes, it is motionEyeOS running on a PI. The PI is connected with my rooter via cable. What more can I do?

ccrisan, 2016-03-15:
Have you set a static IP for your motionEyeOS?

Winki43, 2016-03-15:
Yes. As I had connected my PI via WLAN systime was correct. Didn't try it again but will do on weekend.

ccrisan, 2016-03-15:
When using static IP, make sure you set a correct DNS server. That's the most probable cause for this issue.

closed 2016-07-03 by ccrisan

bojanopatje, 2019-05-12:
Hi, I have the same problem, selected static IP, DNS as it is in terminal...
What I'm doing wrong?

[image]
[image]

jasaw, 2019-05-12:
@bojanopatje Your DNS server is outside of your subnet. Make sure your DNS server address is configured correctly, and your raspberry pi can ping your DNS server at 192.168.160.13

bojanopatje, 2019-05-13:
Hi, thank you for response. The problem was in wifi file that I created following an tutorial from linustechtips. The problem was that I did not change county code.
https://pastebin.com/raw/WCedfA6s
Thank you again and sorry for problems :)