motioneye-project/motioneye #151: OmniVision OV519 not working on motioneye Debian Jessie - transfer drivers from MotionEyeOS

Daniel157, 2016-03-17:
Hey there!

I have a problem with motioneye on Debian Jessie. I installed it according to the instructions here on github, which all worked out well. But when I try to add my webcam to it, I get the message "an error occured, refreshing is recommended". After refreshing, the camera is listed, but not showing a picture. I also noticed, that motioneye isn't even able to start again, if I don't delete the camera before closing motioneye or rebooting the Raspberry Pi.

Normally I would say this camera is just not compatible with motioneye, but what makes me wonder is that I already had this camera properly running when I used MotionEyeOS. I want to use MotionEye now on Debian so that I am able to use the Raspberry also for other stuff.

Is MotionEyeOS using different drivers than Debian Jessie and can I somehow "transfer" them, to make the webcam work?

Here some data about the webcam:

lsusb:
Bus 001 Device 004: ID 05a9:8519 OmniVision Technologies, Inc. OV519 Webcam

hwinfo:
19: USB 00.0: 11200 TV Card
  [Created at usb.122]
  Unique ID: BobO.zGxjQbN6KGC
  Parent ID: ADDn.GVovileGj8F
  SysFS ID: /devices/platform/soc/20980000.usb/usb1/1-1/1-1.2/1-1.2:1.0
  SysFS BusID: 1-1.2:1.0
  Hardware Class: tv card
  Model: "OmniVision USB Camera"
  Hotplug: USB
  Vendor: usb 0x05a9 "OmniVision Technologies, Inc."
  Device: usb 0x8519 "USB Camera"
  Revision: "1.00"
  Driver: "ov519"
  Driver Modules: "gspca_ov519"
  Speed: 12 Mbps
  Module Alias: "usb:v05A9p8519d0100dc00dsc00dp00icFFisc00ip00in00"
  Driver Info #0:
    Driver Status: gspca_ov519 is active
    Driver Activation Cmd: "modprobe gspca_ov519"
  Config Status: cfg=new, avail=yes, need=no, active=unknown
  Attached to: #23 (Hub)

Thanks in advance for any help!

ccrisan, 2016-03-17:
If would be great if we could see some logs. It's either `/var/log/motioneye.log` or the output from `journalctl -u motioneye`.

Daniel157, 2016-03-17:
In `/var/log/motioneye.log` it somehow doesn't produce a log file. But here is the log from journalctl:

-- Logs begin at Thu 2016-03-17 16:29:27 CET, end at Thu 2016-03-17 16:32:37 CET. --
 16:29:34  systemd[1]: Starting motionEye Server...
 16:29:34  systemd[1]: Started motionEye Server.
 16:29:43  meyectl[297]: INFO: hello! this is motionEye server 0.31.3
 16:29:44  meyectl[297]: Illegal instruction
 16:29:44  meyectl[297]: INFO: cleanup started
 16:29:44  meyectl[297]: INFO: wsswitch started
 16:29:44  meyectl[297]: INFO: tasks started
 16:29:45  meyectl[297]: INFO: mjpg client garbage collector started
 16:29:46  meyectl[297]: INFO: server started
 16:31:37  meyectl[297]: INFO: adding new camera with id 1...
 16:31:38  meyectl[297]: ERROR: motion failed to start
 16:31:38  meyectl[297]: Traceback (most recent call last):
 16:31:38  meyectl[297]: File "/usr/lib/python2.7/dist-packages/tornado/web.py", line 1320, in _stack_context_handl
 16:31:38  meyectl[297]: raise_exc_info((type, value, traceback))
 16:31:38  meyectl[297]: File "/usr/lib/python2.7/dist-packages/tornado/web.py", line 1503, in wrapper
 16:31:38  meyectl[297]: result = method(self, *args, **kwargs)
 16:31:38  meyectl[297]: File "/usr/local/lib/python2.7/dist-packages/motioneye/handlers.py", line 235, in post
 16:31:38  meyectl[297]: self.add_camera()
 16:31:38  meyectl[297]: File "/usr/local/lib/python2.7/dist-packages/motioneye/handlers.py", line 158, in wrapper
 16:31:38  meyectl[297]: return func(self, *args, **kwargs)
 16:31:38  meyectl[297]: File "/usr/local/lib/python2.7/dist-packages/motioneye/handlers.py", line 676, in add_came
 16:31:38  meyectl[297]: motionctl.start()
 16:31:38  meyectl[297]: File "/usr/local/lib/python2.7/dist-packages/motioneye/motionctl.py", line 128, in start
 16:31:38  meyectl[297]: raise Exception('motion failed to start')
 16:31:38  meyectl[297]: Exception: motion failed to start
 16:31:38  meyectl[297]: ERROR: 500 POST /config/add/?_=1458228696394&_username=admin&_signature=e89efb0e2ea0032127
 16:31:38  meyectl[297]: ERROR: motion not running, starting it
 16:31:39  meyectl[297]: ERROR: failed to start motion: motion failed to start
 16:31:39  meyectl[297]: Traceback (most recent call last):
 16:31:39  meyectl[297]: File "/usr/local/lib/python2.7/dist-packages/motioneye/server.py", line 326, in checker
 16:31:39  meyectl[297]: motionctl.start()
 16:31:39  meyectl[297]: File "/usr/local/lib/python2.7/dist-packages/motioneye/motionctl.py", line 128, in start
 16:31:39  meyectl[297]: raise Exception('motion failed to start')
 16:31:39  meyectl[297]: Exception: motion failed to start
 16:31:49  meyectl[297]: ERROR: motion not running, starting it
 16:31:49  meyectl[297]: ERROR: failed to start motion: motion failed to start
 16:31:49  meyectl[297]: Traceback (most recent call last):
 16:31:49  meyectl[297]: File "/usr/local/lib/python2.7/dist-packages/motioneye/server.py", line 326, in checker
 16:31:49  meyectl[297]: motionctl.start()
 16:31:49  meyectl[297]: File "/usr/local/lib/python2.7/dist-packages/motioneye/motionctl.py", line 128, in start
 16:31:49  meyectl[297]: raise Exception('motion failed to start')
 16:31:49  meyectl[297]: Exception: motion failed to start
 16:31:53  meyectl[297]: WARNING: Connect error on fd 19: ECONNREFUSED
 16:31:54  meyectl[297]: WARNING: Connect error on fd 19: ECONNREFUSED
 16:31:55  meyectl[297]: WARNING: Connect error on fd 19: ECONNREFUSED
 16:31:56  meyectl[297]: WARNING: Connect error on fd 19: ECONNREFUSED
 16:31:58  meyectl[297]: WARNING: Connect error on fd 19: ECONNREFUSED
 16:31:59  meyectl[297]: ERROR: motion not running, starting it
 16:31:59  meyectl[297]: ERROR: failed to start motion: motion failed to start
 16:31:59  meyectl[297]: Traceback (most recent call last):
 16:31:59  meyectl[297]: File "/usr/local/lib/python2.7/dist-packages/motioneye/server.py", line 326, in checker
 16:31:59  meyectl[297]: motionctl.start()
 16:31:59  meyectl[297]: File "/usr/local/lib/python2.7/dist-packages/motioneye/motionctl.py", line 128, in start
 16:31:59  meyectl[297]: raise Exception('motion failed to start')
 16:31:59  meyectl[297]: Exception: motion failed to start

ccrisan, 2016-03-18:
Well this is either a motion, driver or hardware-related problem. What motion version do you use?

Daniel157, 2016-03-18:
It's version 3.2.12+git2014. I just made sure everything is up-to-date.
I also have the same problem on a "fresh" installation of Debian Jessie. It's weird because everything did work out with this webcam on MotionEyeOS.

ccrisan, 2016-03-18:
Try installing motion-mrdave instead.

Daniel157, 2016-03-18:
I just installed it. I had to make the install myself, I hope I did everything right (used the instructions from the help file of motion-mrdave).
The frontend is still not reachable, the log says the following:

 13:32:42  systemd[1]: motioneye.service holdoff time over, scheduling restart.
 13:32:42  systemd[1]: Stopping motionEye Server...
 13:32:42  systemd[1]: Starting motionEye Server...
 13:32:42  systemd[1]: Started motionEye Server.
 13:32:45  systemd[1]: motioneye.service: main process exited, code=killed, status=4/ILL
 13:32:45  systemd[1]: Unit motioneye.service entered failed state.
 13:32:45  systemd[1]: motioneye.service holdoff time over, scheduling restart.
 13:32:45  systemd[1]: Stopping motionEye Server...
 13:32:45  systemd[1]: Starting motionEye Server...
 13:32:45  systemd[1]: Started motionEye Server.
 13:32:48  systemd[1]: motioneye.service: main process exited, code=killed, status=4/ILL
 13:32:48  systemd[1]: Unit motioneye.service entered failed state.

ccrisan, 2016-03-18:
Make sure to install the indicated ffmpeg package(s) as well.

Daniel157, 2016-03-25:
I installed ffmpeg using `apt-get install build-essential autoconf libjpeg-dev libavformat-dev libavcodec-dev libswscale-dev ffmpeg subversion git`, so I guess that should be fine.

I just noticed that it says the following, when I configure motion-mrdave with the configure command you provide:

pi@raspi-dani:~/motion-mrdave $ ./configure --prefix=/usr --with-ffmpeg=/usr/lib/arm-linux-gnueabihf  --with-ffmpeg-headers=/usr --without-pgsql --without-sdl --without-sqlite3 --without-mysql                                                                                                                              /arm-linux-gnueabihf  --with-ffmpeg-headers=/usr --without-pgsql --without-sdl -                                                                                                                               -without-sqlite3 --without-mysql
**configure: WARNING: unrecognized options: --with-ffmpeg-headers**
checking for gcc... gcc
checking whether the C compiler works... yes
...and so on

When I added the webcam to motioneye, it again wasn't reachable anymore after a reboot of the Raspberry. Log says the following:

16:27:02  systemd[1]: Starting motionEye Server...
 16:27:02  systemd[1]: Started motionEye Server.
 16:27:16  meyectl[324]: INFO: hello! this is motionEye server 0.31.4
 16:27:17  meyectl[324]: Illegal instruction
 16:27:17  meyectl[324]: INFO: cleanup started
 16:27:17  meyectl[324]: INFO: wsswitch started
 16:27:18  meyectl[324]: INFO: tasks started
 16:27:18  meyectl[324]: INFO: mjpg client garbage collector started
 16:27:19  meyectl[324]: INFO: server started
 16:29:48  meyectl[324]: INFO: adding new camera with id 1...
 16:29:49  meyectl[324]: ERROR: motion failed to start
 16:29:49  meyectl[324]: Traceback (most recent call last):
 16:29:49  meyectl[324]: File "/usr/local/lib/python2.7/dist-packages/tornado/web.py", line 1401, in _stack_context_handle_exception
 16:29:49  meyectl[324]: raise_exc_info((type, value, traceback))
 16:29:49  meyectl[324]: File "/usr/local/lib/python2.7/dist-packages/tornado/web.py", line 1603, in wrapper
 16:29:49  meyectl[324]: result = method(self, *args, **kwargs)
 16:29:49  meyectl[324]: File "/usr/local/lib/python2.7/dist-packages/motioneye/handlers.py", line 235, in post
 16:29:49  meyectl[324]: self.add_camera()
 16:29:49  meyectl[324]: File "/usr/local/lib/python2.7/dist-packages/motioneye/handlers.py", line 158, in wrapper
 16:29:49  meyectl[324]: return func(self, *args, **kwargs)
 16:29:49  meyectl[324]: File "/usr/local/lib/python2.7/dist-packages/motioneye/handlers.py", line 676, in add_camera
 16:29:49  meyectl[324]: motionctl.start()
 16:29:49  meyectl[324]: File "/usr/local/lib/python2.7/dist-packages/motioneye/motionctl.py", line 128, in start
 16:29:49  meyectl[324]: raise Exception('motion failed to start')
 16:29:49  meyectl[324]: Exception: motion failed to start
 16:29:49  meyectl[324]: ERROR: 500 POST /config/add/?_=1458923386634&_username=admin&_signature=dfcc0d8d834d3bb4c76cac3878af5431d0b02ccb (192.168.178.132) 515.64ms
 16:29:54  meyectl[324]: WARNING: Connect error on fd 19: ECONNREFUSED
 16:29:55  meyectl[324]: WARNING: Connect error on fd 19: ECONNREFUSED
 16:29:56  meyectl[324]: WARNING: Connect error on fd 19: ECONNREFUSED
 16:29:58  meyectl[324]: WARNING: Connect error on fd 19: ECONNREFUSED
 16:29:58  meyectl[324]: ERROR: motion not running, starting it
 16:29:58  meyectl[324]: ERROR: failed to start motion: motion failed to start
 16:29:58  meyectl[324]: Traceback (most recent call last):
 16:29:58  meyectl[324]: File "/usr/local/lib/python2.7/dist-packages/motioneye/server.py", line 326, in checker
 16:29:58  meyectl[324]: motionctl.start()
 16:29:58  meyectl[324]: File "/usr/local/lib/python2.7/dist-packages/motioneye/motionctl.py", line 128, in start
 16:29:58  meyectl[324]: raise Exception('motion failed to start')
 16:29:58  meyectl[324]: Exception: motion failed to start

Could it be that there is a problem with ffmpeg?

ccrisan, 2016-07-03:
Is this still a problem? If yes, could you try reinstalling everything following [the guide](https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian) as I have updated the ffmpeg version as well as the precompiled motion daemon?

Marijn0, 2026-10-27:
Closing this old issue. If you're still experiencing this problem with recent versions of motionEye and Motion, feel free to open a new issue and include relevant log output.

closed 2026-10-27 by Marijn0