motioneye-project/motioneye #152: Google Drive location?

user150aa346, 2016-03-23:
ccrisan thanks for a cool program. Can you tell me what the syntax is for the Google Drive location? Thanks.

ccrisan, 2016-03-23:
It's the common /path/to/folder syntax. Just make sure that the entire path actually exists on your GDrive.

user150aa346, 2016-03-23:
Just to clarify which is the correct syntax from the picture attached?

/My Drive/Ipsecuritycam/Video
/Ipsecuritycam/Video
https://drive.google.com/drive/folders/xxxxxxxxxxxxxxxxx

I've tried all the variations and still no luck. Thanks.

Regards,
Mike
------------------------------------
Michael A. McKenna
(c) 339-237-2004
I ride to support cancer research:
*You can click this link to donate to my PMC ride
<http://www2.pmc.org/profile/MM0613>*

ccrisan, 2016-03-23:
Please attach the picture here in github.

user150aa346, 2016-03-23:
Sorry no attachment came thru.

Regards,
Mike
------------------------------------
Michael A. McKenna
(c) 339-237-2004
I ride to support cancer research:
*You can click this link to donate to my PMC ride
<http://www2.pmc.org/profile/MM0613>*

user150aa346, 2016-03-23:
sorry, misunderstood what you meant. I've just uploaded to the github.

Regards,
Mike
------------------------------------
Michael A. McKenna
(c) 339-237-2004
I ride to support cancer research:
*You can click this link to donate to my PMC ride
<http://www2.pmc.org/profile/MM0613>*

closed 2016-07-02 by ccrisan