motioneye-project/motioneye #153: Exposure hanging issue with Raspberry Cam

Vortas, 2016-03-23:
Hey there,

I know that issue was already adressed in older topics. 
- Latest motioneye version
- Latest Raspbian version
- Daves motion branch binaries prebuild out of your wiki link

I'm running motioneye on a Pi2 with a Raspberry Cam Board without IR Filter.

So far everything is OK all day long. Image is a bit too bright when sun is at the highest point around noon. So I've put brightness to 45%.

When it's getting dark the light emitted from the street light in front of my house is not enough to get me a image where I can see anything. Ok. Guess the Camera Board is not the best regarding low light...

The next morning after there is daylight again, my image is white. It looks like exposure time didn't set back to lower values. Adjusting the brightness slider 1% is enough. Image jumps back to normal.

I've read in older issues that motioneye is not made to set every camera value and red about using v4l2-ctl to change them. Tried that. Nothing happens.

I've read those posts from 2015 regarding motion and motion-mmal. Looks like Daves branch doesn't include motion-mmal? Tried to get the binary from an older link and switched them. Didn't work. Everything crashed.

So just now I'm stuck and need some help to get around that hanging issue. Today I will get some spare SD cards and install motioneyeos on one and try that.

ccrisan, 2016-03-23:
You haven't enabled autobrightness in motionEye, have you? If you did, disable it. MrDave's motion doesn't have anything to do with the mmal version. If you wish to use the mmal version, the best way to do it right now is to try to install motionEyeOS for your Raspberry PI board variant. The mmal version will automatically be used if you have one single camera which is a RPi camera.

Vortas, 2016-03-23:
You were faster answering me than I could reset my password and add: I'm not using auto brightness, because it does result in a swinging brightness.

I'm just waiting for DHL and the SD cards. Then I will try the OS. 

There DHL is...

Vortas, 2016-03-23:
Done. Now running motionEyeOs. 

And back to the old known problem... I've set a static IP for WLAN. If LAN is unplugged, WLAN stops working. Even after setting WLAN back to DHCP and rebooting... Unplug LAN, Wlan stops working.

ccrisan, 2016-03-23:
motionEyeOS is not designed to work with two network connections. It's either wifi or ethernet. You can temporarily use ethernet to configure the wifi connection, but you then need to boot it with the cable unplugged. Once it detects the ethernet connection at boot, the OS will hang on to that connection and will reboot if it's lost.

Vortas, 2016-03-23:
Unfortunately the WLAN Dongle does not even show activity when rebooting without LAN plugged in. When plugged in it's flashing and there is a wlan0.

Looks like I need to plug a monitor in to see if it's even active without LAN plugged in...

ccrisan, 2016-03-23:
Please do. The latest prerelease version will also print out the wifi driver in use.

Vortas, 2016-03-23:
You might have deleted a part of your sentence? Blind shot of me: lsmod printout

Module                  Size  Used by    Tainted: G
sd_mod                 23572  2
usb_storage            37016  1
scsi_mod               83244  2 sd_mod,usb_storage
spi_bcm2708             4503  0
i2c_bcm2708             4269  0
r8712u                140714  0

ccrisan, 2016-03-23:
I haven't deleted any part of my sentence. You said you'd need to plug in a monitor. I asked to you to "please do". The boot.log should also show what's going on when booting under those circumstances.

Vortas, 2016-03-23:
Sorry. My bad. 

So far: It's loading the right driver. wpa_supplicant needs a while then shows r8712u. That's the correct one. Everything "done".
..
..
..

Starting motioneye: wlan0 has no ip address. 
Then after a while "done"

In the next line follows:
Interface wlan0 has IP address: 192.168.1.132/24
gateway and DNS are correct, too. 

So it's telling me there is a address. But I can't ping it and the WLAN adapter is only flashing for a second while booting. Guess it fetches it's IP then. 

Now for the boot.log... Give me a second to get it.

Vortas, 2016-03-23:
---- booting motionEyeOS 20160320----
 * Mounting filesystems: done
 * Loading kernel modules: done
 * Setting hostname: done
 * Starting syslogd: done
 * Starting watchdog: done
 * Starting brownout detector: done
 * Starting eudev: done
 * Setting root password: done
 * Starting wpa_supplicant: done (r8712u)
 * Configuring wireless network: dhcp
 * Setting current date using http: Wed Mar 23 13:39:34 CET 2016
 * Starting http date updater: done
 * Starting crond: done
 * Starting sshd: done
 * Starting proftpd: done
 * Setting smb admin password: done
 * Starting smbd: done
 * Starting nmbd: done
 * Starting motioneye: done
 # Interface wlan0 has IP address 192.168.1.132/24
 # Default gateway is 192.168.1.1
 # DNS server address is 192.168.1.1

Vortas, 2016-03-23:
Ok slowly it's getting strange. I am able to ping devices in my network from the PI. But I can't ping the PI. It's not even existing with an IP scanner.

Vortas, 2016-03-23:
Ok. I did a complete new installation and did nothing except setting WLAN parameters.

Then I gave up. After work I just put the PI into another room.

And suddenly it works. That's the second time I got MASSIVE problems with WLAN and Raspberry PIs.The same Adapter put into a Laptop has full connection. Put into a RasPi... Connection is not dead but is buggy. Works, Works not, Works partially.

Anyway. It's working next to my Router... 

I will see tomorrow how behavior of brightness hanging with mmal is, when the sun is back.

ccrisan, 2016-07-03:
I guess all the problems went away.

closed 2016-07-03 by ccrisan