motioneye-project/motioneye #155: motioneye.log not created

sgiannulli, 2016-03-25:
Hi,
I'm running 0.31.4 version on raspbian jessie and since version 0.3 motioneye.log file is not created; I'm now trying to solve...any good point where to start from?

many thanks

ccrisan, 2016-03-25:
journalctl -u motioneye

sgiannulli, 2016-03-25:
> pi@raspberrypi:~ $ sudo journalctl -u motioneye
-- Logs begin at Fri 2016-03-25 22:37:56 CET, end at Fri 2016-03-25 22:39:46 CET. --
Mar 25 22:38:14 raspberrypi systemd[1]: Starting motionEye Server...
Mar 25 22:38:14 raspberrypi systemd[1]: Started motionEye Server.
Mar 25 22:38:20 raspberrypi meyectl[356]: INFO: hello! this is motionEye server 0.31.4
Mar 25 22:38:21 raspberrypi motion[607]: [0] [NTC] [ALL] conf_load: Processing thread 0 - config file /etc/motioneye/motion.conf
Mar 25 22:38:21 raspberrypi motion[607]: [0] [NTC] [ALL] config_thread: Processing config file thread-1.conf
Mar 25 22:38:21 raspberrypi motion[607]: [0] [NTC] [ALL] motion_startup: Motion Unofficial-Git-UNKNOWN Started
Mar 25 22:38:21 raspberrypi motion[607]: [0] [NTC] [ALL] motion_startup: Logging to syslog
Mar 25 22:38:21 raspberrypi motion[607]: [0] [NTC] [ALL] motion_startup: Using default log type (ALL)
Mar 25 22:38:21 raspberrypi motion[607]: [0] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
Mar 25 22:38:21 raspberrypi motion[607]: [1] [WRN] [VID] v4l2_select_input: Device doesn't support VIDIOC_G_STD
Mar 25 22:38:23 raspberrypi meyectl[356]: INFO: cleanup started
Mar 25 22:38:23 raspberrypi meyectl[356]: INFO: wsswitch started
Mar 25 22:38:23 raspberrypi meyectl[356]: INFO: tasks started
Mar 25 22:38:23 raspberrypi meyectl[356]: INFO: mjpg client garbage collector started
Mar 25 22:38:23 raspberrypi meyectl[356]: INFO: server started
>

sgiannulli, 2016-03-25:
....also the motioneye.pid is missing from /var/run
so maybe a permission problem?

ccrisan, 2016-07-03:
Raspbian Jessie uses systemd which no longer creates the mentioned log files.

closed 2016-07-03 by ccrisan