motioneye-project/motioneye #156: Take a picture with a USB camera via command line

tajohnsen, 2016-03-25:
First, thank you for this great project.  I'm having a lot of fun playing around with it!
My issue is that I have a USB camera pointing outside and the trees keep setting off the motion detection.  I used area_detect to isolate the trees but for the life of me I can't figure out what command to put after on_area_detected to take a picture with a USB camera..  I'm using motioneyeos on Raspberry Pi 2.

ccrisan, 2016-07-03:
Not possible yet.

ccrisan, 2017-08-05:
This could probably be solved using a mask.

closed 2017-08-05 by ccrisan

supremeliter, 2017-08-06:
Thanks for the reply. I don't need a full explanation but can you elaborate a little on what you mean by mask?

ccrisan, 2017-08-07:
The mask feature has been introduced relatively recently and allows you to define areas that are excluded from motion detection. You can filter out trees or things that move on a regular basis in your camera field of view.