motioneye-project/motioneye #157: How do I set motioneye to continue working when off line

ahollander, 2016-03-27:
When the Motioneye is offline it stops taking time lapse pictures and motion detection stops working. Is there a way to adjust this setting. I'm trying to use the Motioneye remotely.

ccrisan, 2016-03-27:
What do you mean "offline"? How do you know it stops detecting motion?

ahollander, 2016-03-27:
Calin, Thanks for getting back so quickly.  I set up the Motioneye and it takes both still and motion detection video perfectly with Wi-Fi enabled and online. I then move the Wi-Fi out of range (now its offline) to record movement in area not covered by Wi-Fi. when I then bring the Motioneye back into Wi-Fi range it starts taking photos and motion video perfectly again but it never recorded anything when outside of Wi-Fi range. my guess is that it trying to connect to the internet before it runs the rest of the program. if I set it up with Ethernet and then disconnect and move it to the area that I want to record it continues to work perfectly but when I bring it back and reconnect to Ethernet it never reconnects to the internet so I have to do a hard boot, which is obviously risky.

ahollander, 2016-03-27:
to be clear, I move the Motioneye out of Wi-Fi Range

ccrisan, 2016-03-28:
motionEyeOS is not designed to work without a network connection. Sorry.

closed 2016-03-28 by ccrisan

ahollander, 2016-03-28:
Thanks for getting back

a1nosweat, 2017-03-01:
Realise this is a necro, but is there a workaround or alternative solution?  Looking to monitor a parked car away from wifi - Motionpie is brilliant - was also thing of using it for cctv in the garage - but wifi won't reach that far.