motioneye-project/motioneye #159: Not able to detect CSI Camera

Mageschu, 2016-03-28:
Hi, i'm sorry to post a new issue with motioneye and the csi camera module. Motioneye isn't detecting the camera so i can really use some help.

I'm using a Pi 2 with:
- latest version of raspbian with already latest v4l-utils
- latest version of motioneye 
- using prebuild motion branch binaries out of your wiki link
- added bcm2835-v4l2 to the last line to /etc/modules
- enabled camera in raspi-config
- functional csi camera (tested with raspistill)

I can't use motioneyeOs cause there will be other running programms on raspbian after getting motioneye working with the camera.

Any ideas?

Thanks for your help!

ccrisan, 2016-03-29:
See if `bcm2835-v4l2` is in the `lsmod` list. If not, see if you can manually `modprobe bcm2835-v4l2`. Do you have the `/dev/video0` device by the way?

Mageschu, 2016-03-29:
Hi, thanks for helping. I forgot to note: I ran `modprobe bcm2835-v4l2` manually. Without checking `bcm2835-v4l2` in `lsmod`. I'll check the `/dev/video0` device later when i'm home again.

Mageschu, 2016-03-29:
So the bcm2835-v4l2 is in the lsmod list and with ls -ltr /dev/video* i got this: crw-rw----+ 1 root video 81, 0 Mär 28 20:35 /dev/video0

ccrisan, 2016-03-29:
Now I'm puzzled. How do you know motionEye doesn't see this camera? Can I see some screenshots? A log file? An error message, anything?

Mageschu, 2016-03-29:
I made some screenshots / pictures.
[image]
[image]

ccrisan, 2016-03-30:
If you click on Camera Type, do you have the "Local Camera" option?

Mageschu, 2016-03-30:
I don't have any options to choose. 

If i change the port to 80 i'm getting the report, that the connection on port 80 refused.

ccrisan, 2016-03-30:
Can you give me remote ssh access to your setup? There's something fishy going on I think.

Mageschu, 2016-03-30:
I misunderstood your Question, sorry! I thought "Camera" was meant... Sorry :-) Klicking on "Camera type" i'll get two options:
1) Remote motionEye camera
2) MJPEG camera
There's no option called "local camera". Maybe someting went wrong during the installation? I have another Card with a freshly installed Raspbian Jessie... should i try to setup the whole system or do you think that will be wasting time? 
Remote ssh access is still on my to-do list to get to learn but it seems to be a simple task ;-)

ccrisan, 2016-03-30:
I would like to try to debug this setup. We might find a real problem. Can you send me a screenshot of the UI with the settings menu opened?

Mageschu, 2016-03-30:
Here it is
[image]

Mageschu, 2016-03-30:
Whatever happened during the first installation doesn't happened on the second one (other sdcard). It works fine =)

Do you still want to debug the first system? It seems to bei a local and no further existing problem.

ccrisan, 2016-03-30:
Yes i really want to know what went wrong.

ccrisan, 2016-03-31:
If you type `motion -h` on command line what do you get?

Mageschu, 2016-03-31:
Command not found. Before typing the command i became root.

ccrisan, 2016-03-31:
Well have you installed motion? :)

closed 2016-07-03 by ccrisan