motioneye-project/motioneye #164: motion_loop: Video device fatal error - Closing video device

mvberkum, 2016-04-03:
Trying to setup motioneye on Raspberry Pi3, with latest rapbian. Whatever I try, when running motioneye I get the log below (in a loop):

I tried running the motion binary directly on the generated motion config files using 
motion -c /etc/motioneye/motion.conf

No problems then, only when motioneye tries to control it.

Tried both old motion binary and Mr Daves motion, no difference  Suggestions?

```
Apr  2 22:15:30 birdcam2 motion: [1] [WRN] [ALL] motion_loop: Retrying until successful connection with camera
Apr  2 22:15:30 birdcam2 motion: [1] [NTC] [VID] vid_v4lx_start: Using videodevice /dev/video0 and input -1
Apr  2 22:15:30 birdcam2 motion: [1] [NTC] [VID] v4l2_get_capability: #012------------------------#012cap.driver: "raspicam"#012cap.card: "Camera Board OV5647"#012cap.bus_info: "CSI"#012cap.capabilities=0x01000001#012------------------------
Apr  2 22:15:30 birdcam2 motion: [1] [NTC] [VID] v4l2_get_capability: - VIDEO_CAPTURE
Apr  2 22:15:30 birdcam2 motion: [1] [NTC] [VID] v4l2_get_capability: - READWRITE
Apr  2 22:15:30 birdcam2 motion: [1] [NTC] [VID] v4l2_select_input: name = "default", type 0x00000002, status 00000000
Apr  2 22:15:30 birdcam2 motion: [1] [NTC] [VID] v4l2_select_input: - CAMERA
Apr  2 22:15:30 birdcam2 motion: [1] [WRN] [VID] v4l2_select_input: Device doesn't support VIDIOC_G_STD
Apr  2 22:15:30 birdcam2 motion: [1] [NTC] [VID] v4l2_do_set_pix_format: Testing palette YU12 (640x480)
Apr  2 22:15:30 birdcam2 motion: [1] [NTC] [VID] v4l2_do_set_pix_format: Using palette YU12 (640x480) bytesperlines 960 sizeimage 460800 colorspace 00000000
Apr  2 22:15:30 birdcam2 motion: [1] [NTC] [VID] v4l2_scan_controls: found control 0x00980900, "brightness", range 0,100
Apr  2 22:15:30 birdcam2 motion: [1] [NTC] [VID] v4l2_scan_controls: #011"brightness", default 50, current 50
Apr  2 22:15:30 birdcam2 motion: [1] [NTC] [VID] v4l2_scan_controls: found control 0x00980901, "contrast", range -100,100
Apr  2 22:15:30 birdcam2 motion: [1] [NTC] [VID] v4l2_scan_controls: #011"contrast", default 0, current 0
Apr  2 22:15:30 birdcam2 motion: [1] [NTC] [VID] v4l2_scan_controls: found control 0x00980902, "saturation", range -100,100
Apr  2 22:15:30 birdcam2 motion: [1] [NTC] [VID] v4l2_scan_controls: #011"saturation", default 0, current 0
Apr  2 22:15:30 birdcam2 motion: [1] [NTC] [VID] v4l2_scan_controls: found control 0x0098090e, "red balance", range 0,800
Apr  2 22:15:30 birdcam2 motion: [1] [NTC] [VID] v4l2_scan_controls: #011"red balance", default 100, current 100
Apr  2 22:15:30 birdcam2 motion: [1] [NTC] [VID] v4l2_scan_controls: found control 0x0098090f, "blue balance", range 0,800
Apr  2 22:15:30 birdcam2 motion: [1] [NTC] [VID] v4l2_scan_controls: #011"blue balance", default 100, current 100
Apr  2 22:15:30 birdcam2 motion: [1] [NTC] [VID] v4l2_scan_controls: found control 0x08000000, "shutter speed", range 0,65535
Apr  2 22:15:30 birdcam2 motion: [1] [NTC] [VID] v4l2_scan_controls: #011"shutter speed", default 0, current 0
Apr  2 22:15:30 birdcam2 motion: [1] [NTC] [VID] v4l2_scan_controls: found control 0x08000001, "zoom factor", range 1,8
Apr  2 22:15:30 birdcam2 motion: [1] [NTC] [VID] v4l2_scan_controls: #011"zoom factor", default 0, current 1
Apr  2 22:15:30 birdcam2 motion: [1] [NTC] [VID] vid_v4lx_start: Using V4L1
Apr  2 22:15:30 birdcam2 motion: [1] [ERR] [ALL] motion_loop: Video device fatal error - Closing video device
Apr  2 22:15:30 birdcam2 motion: [1] [NTC] [VID] vid_close: Closing video device /dev/video0
Apr  2 22:15:30 birdcam2 motion: [1] [INF] [ALL] motion_loop: Changes:     0 - noise level: 12
Apr  2 22:15:30 birdcam2 motion: [1] [DBG] [ALL] motion_loop: vid_return_code 1

```

mvberkum, 2016-04-03:
Sigh. After spending hours to get to the above state and giving up, I all of a sudden noticed that the log says 'Using V4L1' instead of 'Using V4L2' which is shown when starting motion from the commandline using 'motion -c /etc/motioneye/motion.conf'. Some more searching and googling and I noticed that I missed the 4th bullet on https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian:

- If you want to use the CSI camera module for the Raspberry PI, you need to add bcm2835-v4l2 to /etc/modules and reboot.

Doing that solved it the problem. Confusing that motion worked but motioneye did not, but hey, rtfm ;) Issue can be closed now. Thanks for the great software!

ccrisan, 2016-04-03:
Alright. I'm glad you sorted it out.

closed 2016-04-03 by ccrisan

Marcdup1, 2016-07-12:
@mvberkum I had the same error as you and I fixed it as suggested which worked. After I stop and start motion again it says `Using V4L1`. I can I fix is so that it uses v4l2 all the time for motion?

mvberkum, 2016-07-24:
I would say if you have added it to /etc/modules and have rebooted it should work? Not sure why it didn't for you.