motioneye-project/motioneye #165: Support HEAD HTTP request method

PaulAnnekov, 2016-04-03:
I think that motioneye should support HEAD method. It will allow to detect fast (related to mobile network) if motioneye is still working. E.g. [this](https://play.google.com/store/apps/details?id=com.gadgetjuice.liveimagewidget) mobile app  uses HEAD request to check validate image url and it fails for motioneye :(.

I can't imagine any argument against HEAD support. Even [google.com](http://google.com) supports HEAD.

ccrisan, 2016-04-04:
Alright, HEAD support will be added to motionEye :)

ccrisan, 2016-05-18:
Added with commit 0d78dd0.

PaulAnnekov, 2016-07-02:
Hm, you have added HEAD support only for main page, but not for https://example.com/picture/1/current/ one. And [that](https://play.google.com/store/apps/details?id=com.gadgetjuice.liveimagewidget) mobile app still gets 405 when asking for a HEAD. I think it will be better to add HEAD support for all pages, because HEAD is used to validate that resource exists. I don't see any cons. Should I create new issue?

ccrisan, 2016-07-02:
It's done: d78a9c1cb5bb3ac6802016155399344dfdd09c65 :).

PaulAnnekov, 2016-07-02:
Great man! Now it accepts `HEAD`, but return `Content-Type: text/html; charset=UTF-8`, that's wrong :(. `GET` returns `Content-Type: image/jpeg` and `HEAD` should return the same.

ccrisan, 2016-07-02:
Hmm that means I'd have to implement separate HEAD methods for each possible URIs. I never thought any piece of software written these days would rely on the existence (and the result) of the HEAD method. I guess I'd probably need to properly write HEAD methods for various URIs available.

PaulAnnekov, 2016-07-02:
Yep, `HEAD` is still used. Not so popular, but still...