motioneye-project/motioneye #166: network camera support request

Ethocreeper, 2016-04-04:
i have the ANNKE sp 1 ip cam and when i tri to add it down in the select camera and it says not a supported network camera

can you make an update to support that camera

ozett, 2016-04-15:
do you need username and password for the camera?

or check the right url to use for the annke-cam here:
http://community.ispyconnect.com/ispybb2/viewtopic.php?t=2834

Ethocreeper, 2016-04-17:
i will try 
thanks

Ethocreeper, 2016-04-17:
i did it like the guy in the link you told me. my adress is rtmp://192.168.0.202:7010/live/1jfiegbpyoimq_p0_ZXOTOBRPMANZ but when i put it to motioneye (pi 3b) it says** protocol rtmp not supported**
what do i do now

ccrisan, 2016-05-18:
I don't think the motion daemon supports RTMP. If some versions do support it, please feel free to reopen the issue and I'll make sure the UI accepts it.

closed 2016-05-18 by ccrisan