motioneye-project/motioneye #168: motion log many warnings/errors

hgestel, 2016-04-07:
While evaluating i got a lot of warnings (errors) while the system is working well.

 Non-streaming camera (keep-alive not set)
[2] Content-length present
[1] Non-streaming camera (keep-alive not set)
[1] Content-length present
[1] Non-streaming camera (keep-alive not set)
[1] Content-length present

When it stops collecting the message in syslog is :+1: 
netcam_init_jpeg: no new pic, no signal rcvd

where do I have to look to get the plenty of messages solved?

ccrisan, 2016-07-03:
Yeap I might need to reduce the log level even further as many things are treated as warnings whereas they occur so often that they should be treated as plain info or even debug messages.

ccrisan, 2018-06-20:
This should no longer be a problem with recent motion versions.

closed 2018-06-20 by ccrisan