motioneye-project/motioneye #170: Motion detection not working on rtsp-cams ?

ozett, 2016-04-10:
hi,
i am on motioneye v0.31.4 on debian 8 and have four rtsp-cams. i configured for the first the motion detections, but see nothing happening. is motion-detection possible with your version of motion and rtsp ?

my log gives my an error.
`pr 10 14:40:54 debian82 motion: [3] [ALR] [NET] netcam_handler_loop: netcam camera handler: finish set, exiting
Apr 10 14:40:54 debian82 motion: [4] [ALR] [NET] netcam_handler_loop: netcam camera handler: finish set, exiting
Apr 10 14:40:56 debian82 motion: [0] [NTC] [ALL] conf_load: Processing thread 0 - config file /etc/motioneye/motion.conf
Apr 10 14:40:56 debian82 motion: [0] [NTC] [ALL] config_thread: Processing config file thread-4.conf
Apr 10 14:40:56 debian82 motion: [0] [NTC] [ALL] config_thread: Processing config file thread-5.conf
Apr 10 14:40:56 debian82 motion: [0] [NTC] [ALL] config_thread: Processing config file thread-6.conf
Apr 10 14:40:56 debian82 motion: [0] [NTC] [ALL] config_thread: Processing config file thread-8.conf
Apr 10 14:40:56 debian82 motion: [0] [NTC] [ALL] motion_startup: Motion Unofficial-Git-2caced3 Started
Apr 10 14:40:56 debian82 motion: [0] [NTC] [ALL] motion_startup: Logging to syslog
Apr 10 14:40:56 debian82 motion: [0] [NTC] [ALL] motion_startup: Using default log type (ALL)
Apr 10 14:40:56 debian82 motion: [0] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
Apr 10 14:40:56 debian82 motion: [4] [ALR] [NET] netcam_start: Network Camera thread starting... for url (rtsp://192.168.14.104:554/ch0_1.h264)
Apr 10 14:40:56 debian82 motion: [3] [ALR] [NET] netcam_start: Network Camera thread starting... for url (rtsp://192.168.14.103:554/ch0_1.h264)
Apr 10 14:40:56 debian82 motion: [2] [ALR] [NET] netcam_start: Network Camera thread starting... for url (rtsp://192.168.14.102:554/ch0_1.h264)
Apr 10 14:40:56 debian82 motion: [1] [ALR] [NET] netcam_start: Network Camera thread starting... for url (rtsp://192.168.14.101:554/ch0_1.h264)
Apr 10 14:40:57 debian82 motion: [4] [ALR] [NET] netcam_handler_loop: Camera handler thread [6] started
Apr 10 14:40:57 debian82 motion: [1] [ALR] [NET] netcam_handler_loop: Camera handler thread [7] started
Apr 10 14:40:58 debian82 meyectl[18016]: ERROR: failed to disable motion detection for camera with id 8: Not Valid
Apr 10 14:40:58 debian82 meyectl[18016]: ERROR: failed to disable motion detection for camera with id 5: Not Valid
Apr 10 14:40:58 debian82 meyectl[18016]: ERROR: failed to disable motion detection for camera with id 6: Not Valid
Apr 10 14:40:58 debian82 motion: [2] [ALR] [NET] netcam_handler_loop: Camera handler thread [8] started
Apr 10 14:40:58 debian82 motion: [3] [ALR] [NET] netcam_handler_loop: Camera handler thread [9] started`

how do i debug this? 

i appreciate every hint. thanks..

ccrisan, 2016-04-10:
What is "my" version of motion that you are using?

ozett, 2016-04-10:
i did it strictly like told in the wiki/config manual.. ;-)

> root@debian82:/usr/src# motion -h
motion Version Unofficial-Git-2caced3, Copyright 2000-2005 Jeroen Vreeken/Folker                                                                                                             t van Heusden/Kenneth Lavrsen

usage:  motion [options]

ccrisan, 2016-04-10:
Yeap. That's the best one in my opinion and it does speak RTSP. I have no idea why it doesn't work for you. Ask this on motion-related forums. They may be able to help you.

ozett, 2016-04-10:
ok. thanks.
i saw people reporting working rtsp-configs, but i hoped you had some debugging tips to dive deeper into it.. 
motion is somehow old, development is not that fast, and therefore i will see if i find time

your gui is superb, thanks for that!

ozett, 2016-04-13:
update: 
1) i checked mr-daves for out again:
`git clone https://github.com/Mr-Dave/motion`

2) i installed on my existing debian additionally:
`apt-get install libv4l-dev v4l2loopback-source`

3) i made clean, than compiled like before and as stated in your wiki:
`./configure --prefix=/usr --without-pgsql --without-sdl --without-sqlite3 --without-mysql`

3.1) i stopped motioneye
systemctl stop motioneye

4) i checked for motion detection on command-line

!!! looks like it is running now

!! conclusion: v4l packages had been missing? now will check motion-eye integration..

`./motion -s -c /etc/motioneye/thread-4.conf`

> oot@debian82:/usr/src/motion# ./motion -s -c /etc/motioneye/thread-4.conf
[0] [NTC] [ALL] conf_load: Processing thread 0 - config file /etc/motioneye/thread-4.conf
[0] [NTC] [ALL] motion_startup: Using default log level (NTC) (6)
[0] [NTC] [ALL] motion_startup: Motion Unofficial-Git-2caced3 Started
[0] [NTC] [ALL] motion_startup: Logging to syslog
[0] [NTC] [ALL] motion_startup: Using default log type (ALL)
[0] [NTC] [ALL] motion_startup: Using log type (ALL) log level (NTC)
[0] [NTC] [ENC] ffmpeg_init: ffmpeg LIBAVCODEC_BUILD 3740772 LIBAVFORMAT_BUILD 3740772
[0] [NTC] [ALL] main: Motion running in setup mode.
[0] [NTC] [ALL] main: Thread 1 is from /etc/motioneye/thread-4.conf
[0] [NTC] [ALL] main: Thread 1 is device: rtsp://192.168.14.101:554/ch0_1.h264 input -1
[0] [NTC] [ALL] main: Stream port 8084
[0] [NTC] [ALL] main: Waiting for threads to finish, pid: 26358
[1] [NTC] [ALL] motion_init: Thread 1 started , motion detection Enabled
[1] [ALR] [NET] netcam_start: Network Camera thread starting... for url (rtsp://192.168.14.101:554/ch0_1.h264)
[1] [NTC] [NET] netcam_rtsp_open_context: Using udp transport
[1] [NTC] [NET] netcam_rtsp_resize_ntc:
[1] [NTC] [NET] netcam_rtsp_resize_ntc: ****************************************************************
[1] [NTC] [NET] netcam_rtsp_resize_ntc: The network camera is sending pictures in a different
[1] [NTC] [NET] netcam_rtsp_resize_ntc: size than specified in the configuration file.
[1] [NTC] [NET] netcam_rtsp_resize_ntc: The picture is being transcoded into the size
[1] [NTC] [NET] netcam_rtsp_resize_ntc: requested in the configuration.  If possible change
[1] [NTC] [NET] netcam_rtsp_resize_ntc: netcam or configuration to indicate the same size
[1] [NTC] [NET] netcam_rtsp_resize_ntc: to possibly lower CPU usage.
[1] [NTC] [NET] netcam_rtsp_resize_ntc: Netcam: 640 x 360 => Config: 1280 x 1024
[1] [NTC] [NET] netcam_rtsp_resize_ntc: ****************************************************************
[1] [NTC] [NET] netcam_rtsp_resize_ntc:
[1] [NTC] [NET] netcam_connect_rtsp: Camera connected
[1] [NTC] [ALL] image_ring_resize: Resizing pre_capture buffer to 1 items
[1] [ALR] [NET] netcam_handler_loop: Camera handler thread [2] started
[1] [NTC] [STR] http_bindsock: motion-stream testing : IPV4 addr: 127.0.0.1 port: 8084
[1] [NTC] [STR] http_bindsock: motion-stream Bound : IPV4 addr: 127.0.0.1 port: 8084
[1] [NTC] [ALL] motion_init: Started motion-stream server in port 8084 auth Disabled
[1] [NTC] [ALL] image_ring_resize: Resizing pre_capture buffer to 7 items
[1] [NTC] [EVT] event_new_video FPS 5
[1] [NTC] [ENC] ffmpeg_open Selected Output FPS 5
[1] [NTC] [ALL] create_path: creating directory /tmp/2016-04-13
[1] [NTC] [EVT] event_newfile: File of type 8 saved to: /tmp/2016-04-13/10-40-58.avi
[1] [NTC] [ALL] motion_detected: Motion detected - starting event 1
[1] [NTC] [ALL] motion_loop: End of event 1
[1] [NTC] [EVT] event_new_video FPS 5
[1] [NTC] [ENC] ffmpeg_open Selected Output FPS 5
[1] [NTC] [EVT] event_newfile: File of type 8 saved to: /tmp/2016-04-13/10-41-12.avi
[1] [NTC] [ALL] motion_detected: Motion detected - starting event 2
[1] [NTC] [ALL] motion_loop: End of event 2
[1] [NTC] [EVT] event_new_video FPS 5
[1] [NTC] [ENC] ffmpeg_open Selected Output FPS 5
[1] [NTC] [EVT] event_newfile: File of type 8 saved to: /tmp/2016-04-13/10-41-58.avi
[1] [NTC] [ALL] motion_detected: Motion detected - starting event 3

ozett, 2016-04-13:
help needed:
if i run motioneye, i get no recordings.
if i run it from the commandline with motioneye-config for the cams, i get recodings.

what to check?

ozett, 2016-04-13:
observation:
if i start motion directly in the /etc/motioneye directory, i find the thread files and records.
if i start motion anywhere else in the filetree, it does not find the thread-files.

but i want motion with motioneye and rtsp-recordings. 
something more to check for me? thanks in advance...

ccrisan, 2016-04-13:
Well have you enabled motion detection and movies in motionEye?

ozett, 2016-04-13:
Only Motion, sure!
no Video (i don't need)

ccrisan, 2016-04-13:
Well, do you or do you not need videos?

ozett, 2016-04-13:
sure, i need videos from the motion-detection,
but no daily-video (wich motion could also do).
these are my actual settings in the gui:

[image]
[image]

ccrisan, 2016-04-13:
The movie quality is quite low. Anyways that shouldn't be the cause of your problem. Are you sure there are no videos generated by motionEye? If you click on the camera frame's top right corner, on the movies button, aren't you able to see anything?

ozett, 2016-04-13:
i checked the recording-dir with "watch", no..
and over motion-eyes-gui: no. sorry...thats my problem..

i added the path to the thread-files in motion.conf. did not help.
i made a screenshot of the running processes (with no motion-recording).
looked good, i do not find a clue?!. is there more debuging logs?

[image]
[image]

ozett, 2016-04-13:
must be something with the path, from wich motion is started...

if i start motion (like it is started from motioneye) from the root-dir (with same config), it does not start with the netcam-url.

if i start it from /etc/motioneye-dir, motion is working...

see this:

[image]

ozett, 2016-04-13:
oh, sorry, my mistake. i was to fast, missing the -c config option ...

ozett, 2016-04-13:
1) out of root-dir:
if i start motion alone on the cmd-line,
` /usr/local/bin/motion -n -c /etc/motioneye/motion.conf -d 5`
the same as with the same config-option like it is started with my motioneye 
`systemctl start motioneye'
than i have an error.

it shows not finding the thread.configs for the cams...

2) out of "/etc/motioneye"-directory: success

if i start motion alone on the cmd-line,
` /usr/local/bin/motion -n -c /etc/motioneye/motion.conf -d 5`
the same as with the same config-option like it is started with my motioneye 
`systemctl start motioneye'

it is working...

what to do?

see:
[image]

fenevol, 2016-04-14:
Try to change your motion.conf file, adding absolute path to the thread config file:

`thread /etc/motioneye/thread-1.conf`

ozett, 2016-04-14:
I stated in a post above that I tried that, but wir no success. Motioneye hung than..
There must be another way...
Thanks for the hint..

fenevol, 2016-04-14:
The hint is just to solve the last problem 'Thread config file not found' when starting as a daemon or outside /etc/motioneye. You've tried it for the 'not recording' problem.

Not saving motion detection is another problem you can look at after solving the other one.

ozett, 2016-04-14:
Ok. I can start it from the /etc/motioneye dir.
No Problem. Than I have Motion detection and recordings. But no Web GUI.
I hope to have both. Means normal function for Motion detection inside motioneye. That's why I am bug-hunting ...

ozett, 2016-04-15:
maybe it is related to the "WORKING SCHEDULE"-Feature?

the recording with motion seems to be triggerd through the option
`ffmpeg_output_movies on`

in my camera-config

`/etc/motioneye/thread-7.conf`

and if i start it standalone, the recording is working.

`root@debian82:/etc/motioneye# /usr/local/bin/motion -n -c /etc/motioneye/thread-7.conf`

i dont know how to check logging of motioneye itself, so i cannot see how the working-shedule-feature starts and stops triggering of the recording. maybe it dont starts the recording feature.
can you help me to check this?

sambul13, 2016-04-17:
Make sure you own the dir where you directed Motion to save pics and movies in cam_thread.conf , and comment out same purpose dir in motion.conf .

`sudo chown -R ozett:ozett /home/ozett/Videos/`

ozett, 2016-04-18:
thanks for the hint, but: the motioneye problem persists: 
if i start motioneye for my rtsp-cams, i dont get any recordings..
if i start motion on commandline with the motioneye-config: it works.

as far as i can see, motioneye is running as root via "motioneye.init-debian" in /usr/local/share/motioneye/extra and i have choosen /tmp as recordings dir, because everyone has full rights here, but i dont get recordings via motioneye. 

i appreciate more hints to debug it in motioneye...

> root@debian82:/usr/local/share/motioneye/extra# cat motioneye.init-debian
> !/bin/sh -e
> 
>  -#/etc/init.d/motioneye: Start the motionEye server
> 
> -#BEGIN INIT INFO
> -# Provides:      motioneye
> -# Required-Start: $local_fs $syslog $remote_fs
> -# Required-Stop: $remote_fs
> -# Default-Start:  2 3 4 5
> -# Default-Stop: 0 1 6
> -# Short-Description: Start the motionEye server
> -# Description: Start the motionEye server
> -### END INIT INFO
> 
> NAME="motioneye"
> PATH="/bin:/usr/bin:/sbin:/usr/sbin:/usr/local/bin:/usr/local/sbin"
> DAEMON="/usr/local/bin/meyectl"
> PIDFILE="/var/run/$NAME.pid"
> DESC="motionEye server"
> USER="root"
> OPTIONS="startserver -c /etc/motioneye/motioneye.conf -l -b"
>

sambul13, 2016-04-18:
Did you follow Wiki to install MotionEye for running as USER? Try that and let us know the result. Pls also switch log level to DEBUG in motioneye.conf, restart motioneye server

`service motioneye restart`

from your $Home/user dir, then post the end of motioneye.log here. Without that log its hard to help you, and only you have access to your PC to find it (by using SEARCH).

Motioneye is not that easy to install, so anything is possible. Easy solution may be to remove the package and reinstall & reconfigure.

ozett, 2016-04-18:
i followed the wiki for the debian8 installation,
`https://github.com/ccrisan/motioneye/wiki/Install-On-Debian`
and had no security concerns on my system -> so i let it run as root, like the wiki said for debian8  ....

i set loglevel to debug an got a lot of information...
i did this with motioneye (via motioneye.conf) and again as root with motion from the commandline for my motioneye-thread4.conf (one of my rtsp-cameras).

can i attach the file here? it is really long..

sambul13, 2016-04-18:
You can paste any text at [Pastebin](motioneye/thread-7.conf), and post here the link. Pls also paste separately the content of your motioneye/thread-7.conf or similar config file for your RTSP cam. Change cam IP and password for security.

ozett, 2016-04-18:
ok, thanks for reply. i got hope...

after googling i found it easy to paste from the commandline to termbin, so here here is my thread4-conf defintion from motioneye.

thread-4.conf (http://termbin.com/pgcg)

the log while using motioneye (no recordings. i promise: i moved in front of the cam!)

http://termbin.com/yg67

the log while using motion from commandline (succes..)

http://termbin.com/1yf6

sambul13, 2016-04-18:
@ozett 

The logs you posted don't represent equivalent set of data time and content wise, so its hard to pinpoint the error. Try deleting the log file for a select cam, then test it in both above approaches for 20-30 sec each, then post the entire log. I wonder if Motion creates its own separate log for the cam you can post? Of course, _ccrisan_ might just point the cause more easily.

I also want to point, you seems to misconfigure several cams like that:

```
The network camera is sending pictures in a different
size than specified in the configuration file.
The picture is being transcoded into the size 
requested in the configuration.  If possible change
netcam or configuration to indicate the same size
to possibly lower CPU usage.
Netcam: 640 x 360 => Config: 1280 x 720

```
Then you set motion detection _threshold 27648_ in your cam config example, so it may be too high for the small reso cam. But again, it would reflect in both attempts.

ozett, 2016-04-18:
hi,
i really appreciaty your reply. thanks a lot, it keeps me going here..

i will create two more logs, one from motioneye and one from motion and post here.

for the misconfiguration and treshhold: your are right, but it makes me wonder that the same configuration from motioneye is working standanlone in motion. ...!?!?

i come back in some minutes with the logs...
(big thanks again.)

sambul13, 2016-04-18:
I looked at a generic motioneye cam config again, and it appears to resemble similar Motion cam config, but only without commented text. So now I think Motion may indeed start standalone by using motioneye cam config file. But the Q is, why do you need to start Motion standalone instead of only using motioneye, since it starts motion daemon automatically.

So try to make sure, standalone Motion daemon is set to OFF in motion.conf before you test.

ozett, 2016-04-18:
i found i comandlinetool to paste it, but i have no api_dev_key configured for pastebin. it went per default to debian. but it is accessible.

the motion.log from motioneye: 
(where i jumped in front of the cam, but no motiondetection. ...)

http://paste.debian.net/439196/
http://termbin.com/o4mq

and the same thread-4.conf from motioneye, but called standalone with:
` /usr/local/bin/motion -l /tmp/motion-thread4.log -n -d 9 -c /etc/motioneye/thread-4.conf`

http://termbin.com/nsgd

-------------------
i dont know, if i got the point from your last posting right.
i start motion only standalone for testing purpose. i use -n (no daemon) in the command line.
i use the thread-4.conf and ommit the motion.conf, because of some @embbeded configs not realy working for me in this test setting. (is that the important part?)
i do this only, because motioneye is not recording. i dont know why.
so both variants share the same and only configuration, which is written through motioneye.

sambul13, 2016-04-18:
I suggested to use only motioneye service, but not motion service, since you wrote above:

```
1) out of root-dir:
 if i start motion alone on the cmd-line,
 /usr/local/bin/motion -n -c /etc/motioneye/motion.conf -d 5 
 the same as with the same config-option like it is started with my motioneye 
 `systemctl start motioneye'
 than i have an error.
it shows not finding the thread.configs for the cams...
2) out of "/etc/motioneye"-directory: success
if i start motion alone on the cmd-line,
 /usr/local/bin/motion -n -c /etc/motioneye/motion.conf -d 5 
 the same as with the same config-option like it is started with my motioneye 
 `systemctl start motioneye'
it is working...

```
Onviously, the question is, why do you need to start Motion in the 1st place, when using motioneye, since it starts motion daemon automatically?

Can you now post your motion.conf and motioneye.conf stored in motioneye dir?

ozett, 2016-04-18:
> ""Obviously, the question was, why do you need to start Motion standalone when using motioneye, since it starts motion daemon automatically with correct permissions?""

THIS is the question. 
BUT i dont want to start motion alone.
BUT if it starts with motioneye, i cannot see, why i dont get recordings.

Obiously both ways look the same. at the outside..

--------------------------------------------
the systemctl thing: 
i use TAB for cmd-line completion. sometimes it i do it, the completion results 

`systemctl start motioneye.service`

and sometimes i put in via keyboard

`systemctl start motioneye`

as i did this so after the wiki for motioneye and i was newly pushed to debian 8 -- 
i considered this as equal options. a mistake?

---------------------------------
permissions on the video-dir 
(i did it before with /tmp, which has more permission for all. without difference to the new dir:)

> root@debian82:/etc/motioneye# sudo ls /mnt/nfs-motion/wohnzi -al
> insgesamt 20
> drwxrwxrwx  4 root root 4096 Apr 18 08:15 .
> drwxrwxrwx 17 root 1004 4096 Apr 16 21:03 ..
> drwxr-xr-x  2 root root 4096 Apr 18 17:22 2016_04_18
> drwxr-xr-x  2 root root 4096 Apr 14 21:56 ants01-2016-04-14
> -rw-r--r--  1 1003 1004   69 Mai 10  2015 ftp.txt
> -rw-r--r--  1 root root    0 Apr 14 21:13 .keep
> root@debian82:/etc/motioneye# ^C

motion.conf:

```
> root@debian82:/etc/motioneye# cat motion.conf
> # @enabled on
> # @show_advanced on
> # @normal_password
> # @admin_username admin
> # @admin_password
> # @normal_username user
> 
> 
> thread thread-4.conf
> webcontrol_html_output on
> webcontrol_port 7999
> setup_mode off
> webcontrol_localhost on
> root@debian82:/etc/motioneye#
```

motioneye.conf

```
root@debian82:/etc/motioneye# cat motioneye.conf

# the base URL path at which motionEye lives
# change this if you run motionEye behind a reverse proxy (e.g. nginx),
# and you want to make motionEye accessible at a specific base path (e.g. /cams)
#base_path /cams

# path to the configuration directory (must be writable by motionEye)
conf_path /etc/motioneye

# path to the directory where pid files go (must be writable by motionEye)
run_path /var/run

# path to the directory where log files go (must be writable by motionEye)
log_path /var/log

# default output path for media files (must be writable by motionEye)
media_path /var/lib/motioneye

# path to the motion binary to use (automatically detected if commented)
#motion_binary /usr/bin/motion

# the log level (use quiet, error, warning, info or debug)
#log_level info
log_level debug

# the IP address to listen on
# (0.0.0.0 for all interfaces, 127.0.0.1 for localhost)
listen 0.0.0.0

# the TCP port to listen on
port 8765

# whether motion HTTP control interface listens on
# localhost or on all interfaces
motion_control_localhost true

# the TCP port that motion HTTP control interface listens on
motion_control_port 7999

# interval in seconds at which motionEye checks the SMB mounts
mount_check_interval 300

# interval in seconds at which motionEye checks if motion is running
motion_check_interval 10

# interval in seconds at which the janitor is called
# to remove old pictures and movies
cleanup_interval 43200

# timeout in seconds to wait for response from a remote motionEye server
remote_request_timeout 10

# timeout in seconds to wait for mjpg data from the motion daemon
mjpg_client_timeout 10

# timeout in seconds after which an idle mjpg client is removed
# (set to 0 to disable)
mjpg_client_idle_timeout 10

# enable SMB shares (requires motionEye to run as root)
smb_shares false

# the directory where the SMB mount points will be created
smb_mount_root /media

# path to the wpa_supplicant.conf file
# (enable this to configure wifi settings from the UI)
#wpa_supplicant_conf /etc/wpa_supplicant.conf

# path to the localtime file
# (enable this to configure the system time zone from the UI)
#local_time_file /etc/localtime

# enables shutdown and rebooting after changing system settings
# (such as wifi settings or time zone)
enable_reboot false

# timeout in seconds to use when talking to the SMTP server
smtp_timeout 60

# timeout in seconds to wait media files list
list_media_timeout 120

# timeout in seconds to wait for zip file creation
zip_timeout 500

# timeout in seconds to wait for timelapse creation
timelapse_timeout 500

# enable adding and removing cameras from UI
add_remove_cameras true
root@debian82:/etc/motioneye#
```

sambul13, 2016-04-18:
Looking at capture logs:

[```
1] [DBG] [ALL] [Apr 18 08:48:20] motion_loop: post capture 0
[1] [INF] [ENC] [Apr 18 08:48:20] ffmpeg_avcodec_log: AVFrame.format is not set
 - flag 24
[1] [INF] [ENC] [Apr 18 08:48:20] ffmpeg_avcodec_log: AVFrame.width or height is not set
 - flag 24
```

The above appears when motion is trying to save the frame. However, in the 1st log it never tried to save any frames. This might be ffmpeg permissions issue.

So try that:
- stop motioneye service and forget about motioneye
- setup cam4.conf in Motion own config dir
- set log to debug in motion.conf in Motion own dir
- start Motion standalone with its own config ( not with motioneye conf)
- post motion.log from Motion own dir (not motioneye.log or motion.log from motioneye dir)

ozett, 2016-04-18:
how will this help me to get motioneye running with recordings?
i would copy most settings from thread-4.conf to another dir, give this config to motion and will post the debug. but is there a great difference to run it standanlone with my commandline?

seems to me at the moment, that 
1) my ffmpeg is broken (when running with motioneye...)
2) i need help from motioneye developer.

but i will undergo creating another debug log. thanks for accompanie... hope to get this nailed down, motioneye is great..

sambul13, 2016-04-18:
It will help in showing whether Motion works normally with its dependencies from its own dir apart from motioneye. Keep in mind, there is difference btw motion.conf stored in motion dir and motioneye dir, so you'll be better off creating motion confs rather than copying them.

ozett, 2016-04-19:
i think i didn get the point...
Motion IS working OK. it must be the same motion binary, that motioneye is using. i have only this one on my system.
Motion is working OK, it records motion with the confs for the cam from motioneye.

why install motioneye again? i assume that motioneye only starts this one and only motion binary.
but maybe it triggers not the motionrecording feature somehow. maybe because of the working schedule settings? i cannot dive deeper into motioneye sourcecode, so i was hoping for help from the dev. it is not clear to me, why i should re-install motioneye. i assume i would not change anything of the motion binary. as the wiki states, after compiling, motion binary is the single thing that is needed for motioneye to watch motion...

ozett, 2016-04-19:
you pushed my again in some direction ...  :smiley:

i looked up the source and found the calling of the motion detecion in motioneye here:

https://github.com/ccrisan/motioneye/blob/f613c937820bffee09698026350f4bddd9e79f4c/motioneye/motionctl.py#L214

i checked this in my running motioneye instance, and it says, it is not running.
now i will check, if the url come from the motion binary, and why it says, motiondeteciton is NOT running, even if i configured it to be running in the webgui..

[image]

sambul13, 2016-04-19:
Assuming its not a bug, most viable reason is broken permissions (no write access) of certain folders and files possibly ffmpeg related, hence no record in the log. That's why its much easier to reinstall motioneye, if Motion standalone works OK with ffmpeg. :)

ozett, 2016-04-19:
SOLVED

Hey, hey, thanks, its running...

:blush:

after your suggestions i had only one cam running.
now diving deeper into the inner workings to see how dection is triggered, i discovered the http-api of motion. to see what was going on, i could only access it fromlocalhost on port 7999.

i changed one thing in /etc/motioneye/motion.conf:

`wecontrol_localhost off`

what was before "on", to access the webcontrol from my PC 
and see all statistics for the running motion instance in my browser. 

it documented here:
http://www.lavrsen.dk/foswiki/bin/view/Motion/MotionGuideAlphabeticalOptionReferenceManual#control_port

but it is called "control_port" instead of "**web**"control_port (in motioneye), and accesses the http-api of motion:
http://www.lavrsen.dk/foswiki/bin/view/Motion/MotionHttpAPI

i assume now, that motioneye had no access to the local instance of webcontrol with 127.0.01:7999
dont know why, but opening the api for acess from the local lan helped.
i have no security concerns, like before. 

i will see if i have to close it up again and reproduce the error, and see, what must be changed,
but maybe the dev reads here and looks also into this...

thanks alot for helping me out ...

:blush:

ozett, 2016-04-19:
i stops working somehow, if i change something in the webgui of motioneye, 

i can go to the webcontrol of motion and check the detection status.
i says, it is in pause mode. if i change manually to active, i see me red square for motion-detection again in the motioneye webgui.

i will observe how it is running further...

sambul13, 2016-04-20:
Detection works for me without working schedule activated. At least I didn't activate it manually when adding a cam, but detection worked. It seems to be activated by default in motioneye. See also:
[Schedule / Enable / Disable Detection with Cron](http://superuser.com/questions/169654/how-to-schedule-motion-detection)

ozett, 2016-04-21:
i observed, that after a configuration change the detection status is in pause for the specific cam.
i check this on the http-api of motion: http://192.168.14.212:7999/2/detection/status
must be an motioneye issue...

ccrisan, 2016-04-21:
@ozett you probably have Working Schedule enabled in motionEye.

ozett, 2016-04-21:
no. 
i checked this for sure, as i debuged my issue here longer....
i can check this again, if it is reproducable. but i assume (after all) it is.
could also be a motion-issue, that i changes to "pause",
but i watched this carefully, so i think it comes after re-configuration of the cams via motioneye..

ccrisan, 2016-04-21:
Your thread config file (attached above a few days ago) has the following line:

    # @working_schedule 00:00-00:00|00:00-00:00|00:00-00:00|00:00-00:00|00:00-00:00|00:00-00:00|00:00-00:00

Remove that line and you should be fine. I'm not sure why the UI didn't remove the line itself.

ozett, 2016-04-21:
Thanks for reply, 
the Logs are from the first test Cam,
i am now on a another Cam.
But the cam is in detection Pause state after configuration via motioneye. 
As soon as I am back to my system I will check, if I can reproduce this an will Report the details...
Thanks until now

ccrisan, 2016-04-21:
@ozett then please attach your new config files (all of them, so I can reproduce the issue myself) and check whether your working schedule is enabled at all in UI.

sambul13, 2016-04-21:
@ccrisan 
How to comment out select lines in motioneye.conf and motion.conf used by motioneye? They seems to already use # for a different purpose.

ccrisan, 2016-04-21:
Just remove the @ and it will be treated as a simple comment.

ozett, 2016-04-21:
i am at my system.
i reconfigured the Frame Change Threshold on my (at the moment: one and only) rtsp-detection cam, than went to the http-api of motion and checked: http://192.168.14.212:7999/2/detection/status

it was than paused.

in the motioneye gui i did not change anything else, the status before was active. 
the camera recorded over the day two movies.

motion.conf  : http://termbin.com/fhz4
thread-7.conf  :  http://termbin.com/ie7g

i now activate the cam with the http-api, because it still is active in the motioneye-gui.
my you give me another hint..?

sambul13, 2016-04-21:
@ccrisan 
Is it possible to switch motion detection on/off manually in motioneye for a select cam, when Working Schedule for this day is set to On?

Is it possible to set Motion Detection on/off for all cams at once? How I can set 5 min delay with such command? Its required, for example, when leaving office.

ozett, 2016-04-21:
i came back to my system to check some recordings, but meanwhile the status or motion-detection changed to pause for that specific cam. maybe motion does this by itself on some conditions (i dont know), or is it motioneye?....

ozett, 2016-04-21:
i stopped and started motioneye, to get a new and clean motion.log, 
and the (one and only rtsp) cam was in paused mode (for recording) ...
how comes this? i started the cam manually again via http-api...

sambul13, 2016-04-26:
Did you manage to fix the issue? I now experience it too, with ANY cam type. Can't find what's wrong based on configs. The logs don't say much either. I started a dedicated thread [here](https://github.com/ccrisan/motioneye/issues/193), since its not restricted to RTSP cams.

ozett, 2016-04-26:
i have the one and only testcam with rtsp running.
i cannot find, why detection is going to pause a some point in time....
the logs of motion itself are not saying much, you are right...

i will see, if i can debug the status of motion-detection in the http-api of motion....
could also be a task for a watchdog/status-button in motion-eye..
or motioneye could log the status on its own...

if i find something more, i will report. if you like ..

sambul13, 2016-04-26:
Of course if you find the cause report it, it will help the developer. I mentioned some presumed causes in the above linked thread.

sambul13, 2016-04-26:
I tried to move motion bin from /usr/local/bin/motion to /usr/bin/motion in a hope it would fix the issue, but it isn't. Some setting in Motioneye switches detection to PAUSE every time a change made in any cam settings in motioneye webui. [See this pic](http://s32.postimg.org/vzw1ittxh/motioneye_detection.jpg). Basically, usability of motioneye is minimal with this bug, since a secondary interface is required to control motion.

ozett, 2016-04-27:
cool, thanks for providing another proof of this behaviour..
sometimes one sits alone on his computer and thinks one is only oberving a singular phenomenon,
and another proof prevents for getting somekind of crazy... :neckbeard:

sambul13, 2016-04-27:
Can you suggest why this error may happen each time APPLY button is pressed in motioneye webui to change settings of any cam?

[8 more replies]

closed 2026-11-07 by JamBalaya56562