motioneye-project/motioneye #171: Invalid client_id with Dropbox in motionEye 0.31.4

kanschat, 2016-04-10:
Hi, 

I am using motionEye 0.31.4 on a Raspberry Pi 2 and I wanted to use the Dropbox-Upload Service. When I am clicking "Obtain Key" an new Browser-Tap opens with the URL: 

`https://www.dropbox.com/1/oauth2/authorize?response_type=code&client_id=dropbox_client_id_placeholder`

I think the placeholder should be replaced with an ID.

When I use Google Drive the "Obtain Key"-Button is working correctly.

ccrisan, 2016-04-10:
How did you install motionEye?

kanschat, 2016-04-10:
I followed the instructions in the wiki (0.29). Since then I upgraded with sudo pip install.

ccrisan, 2016-04-10:
Please use [these](https://github.com/ccrisan/motioneyeos/blob/master/package/motioneye/dropbox.keys) or register your own app on DropBox. The DropBox guys have asked me to remove those from motionEye's source code for "security" reasons, although there's no other way around it :(

ccrisan, 2016-04-10:
My apologies, I forgot to tell you where to use them. Please edit the /usr/lib/python2.7/site-packages/motioneye/uploadservices.py file (or a similar location) and look for `dropbox_client_id_placeholder`. You'll figure it out. Yes, it will be overwritten with the next upgrade. No, there's no other way to do this, that I know of.

CrazyEyesPete, 2016-04-22:
I was also having this problem with a Pi server. Thanks for this question @kanschat and your answer @ccrisan

CrazyEyesPete, 2016-04-22:
I found my uploadservices.py file at /usr/local/lib/python2.7/dist-packages/montioneye

CrazyEyesPete, 2016-04-22:
OK, Now that I've found the file and where to replace the IDs they are the same in my local file as the one provided in your above link.

What are my next steps?

CrazyEyesPete, 2016-04-22:
After obtaining an authorization key from DropBox I get an  error in MotionEye stating:  **'Accessing the upload service failed: 'Dropbox' object has no attribute '_refresh_credentials'!**

I'm running on Raspian with MotionEye ver 0.31.4

ccrisan, 2016-04-23:
@CrazyEyesPete can I see your motionEye.log file after such an error?

ccrisan, 2016-05-18:
8f38d96 should fix this error.

closed 2016-05-18 by ccrisan

PhilipWhiteside, 2019-12-19:
@ccrisan I'd suggest a comment in the UI for this, as it threw me at first, and I'm sure it will many others. I'm on `0.42` docker version and did not see any warning around Dropbox not working. Knowing I don't have to do this workaround for GDrive each update has prompted me to switch over to that.

nomadic-squirrel, 2019-12-31:
I updated my Docker image so I'm running 0.42, but when hitting the Obtain Key link, a new tab opens to a url that seems to contain the client_id, but then redirects to the same place holder error:

[image]

Previously, on 0.41, it gave the same error, but without the redirect.

Ormerod2, 2020-02-22:
Same issue here. Running motioneye in Docker image 0.42

kestena, 2022-01-06:
I found a way to do it. 

Go to this link from ccrisan:
https://github.com/ccrisan/motioneyeos/blob/master/package/motioneye/dropbox.keys

You will find the keys to Dropbox there.

Now go to your raspberry pi, and locate the uploadservices.py file. Some find it under /usr/local/lib/python2.7/dist-packages/motioneye/uploadservices.py, I personally found mine under docker since I run my version under docker.

If you use docker locate the file using `sudo find . -name uploadservices.py `.

The path will be something like docker/use/lib/overlay2/somenumbersandletters/usr/lib/etc.... 

I stopped the docker container first before attempting the changes in the uploadservices.py file. To do that just list the container under docker with `docker container ls`

Then copy the motioneye container (letters and number). Stop the container: `docker container stop [name of container]`

Nano to uploadservices.py, scroll until you find the Dropbox section and exchange the "dropbox_client_id_placeholder" with the first client_id from the Github link I pasted in the beginning. 

Do the same for the second setting there named: NOT_SO_SECRET something. Exchange the parameter with the second parameter from the Github link. Save with CTRL X, Yes, Enter. Restart the docker container with `docker container start [name of container]` and you are done here.

Go to the Motioneye settings, under Dropbox now it should work!

Hopefully I was able to help you.

Regards,
mislishsezahaker