motioneye-project/motioneye #172: Artifacts on screen while streaming

Massgun, 2016-04-11:
Hi,

first of all I would like to thank you for this easy to use piece of software. I am using it since yesterday evening and am very impressed :)

I have a question tho, when I am watching the live streams on my computer via the link to the Pi. The longer the connection is open the more 'artifacts' are shown. When I logout and login again there is no change. When I change the resolution or quality it's good for a limited amount of time but the artifacts will return. 

I added a picture to show you what I mean. 
[image]

How can I fix this? Or is it maybe camera related?

ccrisan, 2016-04-11:
Please open this issue on a motion-related forum. When problems like this occur, there's nothing motionEye can do about it. Sorry.

closed 2016-04-11 by ccrisan

Massgun, 2016-04-11:
Ok, thank you anyway :)