motioneye-project/motioneye #173: Failure to get wireless IP results in reboot loop

olavt, 2016-04-11:
I have setup motioneye on a Raspberry Pi 3 to use WLAN, but for some reason there is an error in the WLAN config. This results in the following messages:

...network: wlan0 had no IP address for 20 seconds, rebooting

I have connected it to a wired network. However, since a reboot is triggered in a loop it's a bit hard to logon and fix the issue. I also normally have no display connected, so it's hard to figure out what's going on.

What is the best way to recover? Why couldn't motioneye just be happy with the IP from the wired conection (eth0) and work with that IP?

ccrisan, 2016-04-11:
Once a wifi connection has been configured, motionEyeOS insists on using that as a main connection. If it goes down, motionEyeOS will reboot in hope that the wifi connection will be fixed after reboot.

The question is how the heck does your wifi IP address get lost? Does anything happen to the router in the meantime? Could you please attach all of your log files so I can look through them?

olavt, 2016-04-11:
One scenario. I connect the headless Raspberry Pi 3 to a LAN port on a router in order to setup for the first time. I logon to motionpie to configure WLAN. What if I enter some error in SSID or the password. When you click Apply it reboots. If it then fails to get to the WLAN it will go into the reboot loop. Another scenario. I setup and test everything on my home WLAN. Then I bring the device to my cottage with the different WLAN and it will also just reboot and I will not be able to get into it via the LAN, since it just reboots.

ccrisan, 2016-04-12:
I agree with you - there should be a way to recover from a wifi problem using an ethernet connection, at least.

bizquick3, 2016-04-12:
I am experiencing this issue as well on a RPi2. Maybe just as a quick fix for the time being, the number of reboots could be limited for a given time period? After 3 reboots in 5 minutes the system could hold off on rebooting for a set time period? Just a suggestion. Thanks for all your hard work on this project!

ccrisan, 2016-04-12:
Just to be clear, is this caused by a wifi misconfiguration caused by you or the OS wifi script fails for some unknown reason (and we still need to investinage what is going on)?

bizquick3, 2016-04-12:
I don't believe its a wifi misconfiguration, if I do a physical power cycle it usually connects the first time without any problems. Most of the time this problem happens when I trigger a reboot through the web interface. I'll try to reproduce the problem and attach some log files for you. This could even be something as simple as the wifi signal being too weak, or a weak power supply. I'll do some more experimenting and update you when I get a chance.

ccrisan, 2016-04-12:
Please do. In the next version I will make sure the ethernet connection is used if available.

ccrisan, 2016-06-05:
Fixed with recent commits.

closed 2016-06-05 by ccrisan

giuseppepirozzi, 2018-04-12:
hello ccrisan, i have the same problem. 
The problem was created because the disk space saturated. 
Now i have freeed the space mounting the image on another linux version and cancelling the mpeg files.
But the problem is not solved.
I tryed the wired connection but it reboot the same.
How can i disable manually the wifi connection ?

I don't wish to rewrite the image because i did some customizations.

thanks.
bye

ccrisan, 2018-04-13:
@giuseppepirozzi you can disable wireless networking by removing the network block from the  `/data/etc/wpa_supplicant.conf` file.

giuseppepirozzi, 2018-04-13:
thanks @ccrisan ccrisan at the end, i rewrited the card......