motioneye-project/motioneye #174: Saving to animated GIFs for email notifications

sambul13, 2016-04-13:
Is it possible to add such unique feature to MotionEye? To my knowledge, no Linux package allows to save a series of motion detected jpegs to an animated GIF before attaching it to Detected Motion Email Notification. On Windows, only [ContaCam](http://www.contaware.com/contacam.html) can do that, which saves and emails small anim GIFs in addition to larger MP4 videos of motion detected events.

Emailing anim GIFs is much more suitable that a series of JPEG snapshots. Anim GIFs can be easily created with FFMpeg, ideally with a running Date & time stamp overlay like in ContaCam.

At least two approaches can be use for that:

1. Convert a series of JPEG snapshots to a small anim GIF like [here](https://community.linuxmint.com/tutorial/view/1118)
2. Convert just captured and saved MP4 video to anim GIF like [here](http://mwholt.blogspot.ca/2014/08/convert-video-to-animated-gif-with.html). Higher quality can be achieved with [this](http://blog.pkh.me/p/21-high-quality-gif-with-ffmpeg.html) technique.  

Once generated, the anim GIF can be sent by Motion with a notifiication email instead of JPEG images. This will be a killer feature for MotionEye, distinguishing it from other surveillance packages. :)

ozett, 2016-04-18:
maybe it can be done like the image-crop with imagemagick in the solution here:

https://github.com/ccrisan/motioneye/issues/179

sambul13, 2016-04-18:
Thanks for the useful hint. Ideally it should be possible to setup through motioneye webUI, but I'll try that approach too. ImageMagick is definitely a good tool to generate animated GIFs.

Another similar approach to generating animated GIFs would be generating both a full reso video of a motion detection event, and a timelapse video of the same event. Then the timelapse video can be sent with email notification, while full video will be kept locally. Unfortunately, while Motion allows to generate both video types, it doesn't allow to do it in parallel, and such option is absent in motioneye as well.

Can someone suggest a small script that would do it? Also, would be nice if motioneye added such option to the webUI, since sending a series of still images or a large video with email notifications is not practical and has limited usability.

sambul13, 2016-04-18:
If an H264 cam can output 2 or more streams with different resolution like 720P and CIF/SIF(625) 352 × 288, it makes sense to use a low resolution stream for motion detection and email notification movies, and its high resolution stream for database recordings. Both audio:video recording types can be done by ffmpeg straight from the source without transcoding instead of using analyzed by Motion jpeg pictures for video recordings. IMHO its a much cleaner approach, and very low PC CPU taxing.  I wonder if its possible to add such feature to motioneye, despite it riches beyond current Motion feature scope?

ccrisan, 2016-07-03:
H264 support is on the way, through motion, however. I still need to find a fix for some Raspberry PI-related issues and recent ffmpeg versions. The GIF thing won't happen.

closed 2016-07-03 by ccrisan

rexfm, 2016-07-13:
I was able to get simple animated gifs working with this little hack:
I installed "gifsicle" b/c it's way faster at generating gifs on the RPi and then added this one line command to the "File Storage" -> "Run a command" camera setting:
`ffmpeg -i %f -vf scale=320:-1 -r 1 -f image2pipe -vcodec gif - | gifsicle --delay=10 -O2 --multifile --loop - > %f.thumb`

More details here: http://rex.fm/auto-generate-animated-gifs-from-security-camera-footage-with-motioneye-hack/

fireheadman, 2016-07-16:
rexfm... left you a comment on your site.   I am on ver 0.32.2 (fresh build on jessie w/motioneye).  Using your hack created the *.thumb, but I do not have animated gifs to see.