motioneye-project/motioneye #175: Controls for contrast, brightness

maxholgasson, 2016-04-14:
Hi, is there any chance to add controls for contrast, brightness etc. That would be great as my night images are quite soft

ccrisan, 2016-04-15:
Contrast and brightness are already there if your camera supports them. What camera do you use?

maxholgasson, 2016-04-15:
Hi,
i'm using an analog camera with an EasyCap Video USB Grabber Fushicai UTV007

ccrisan, 2016-04-15:
What's the output of the following command:

    v4l2-ctl -d /dev/video0 --list-ctrls

maxholgasson, 2016-04-15:
mh, it doesn't display anything. no output at all

ccrisan, 2016-04-17:
Well if there's no output to that command, motionEye will consider that your video device has no controls at all. I'm sorry but there's nothing motionEye can do to implement brightness/contrast controls for your camera.

closed 2016-04-17 by ccrisan