motioneye-project/motioneye #177: feature-request: ordering of CAMs

ozett, 2016-04-15:
hi,
i added another cam. it is added at the end of the existing list, and the preview of the cam is inserted between the preview of my other cams. i would like to control the ordering of the preview windows.
may you could add some numbering/ordering option to control the cam ordering in the preview window?

cool, thanks..

ccrisan, 2016-04-15:
It's something I've been planning on implementing. There's no ETA for this but it's definitely going to happen.

electron1979, 2018-12-10:
Looks like this is done. Close???

atomicwrites, 2021-04-10:
@electron1979 As far as I can tell, this is not done. You have to stop motioneye and edit the configs manually. Or was something else added?

map7, 2022-10-17:
I've written a little script to help move cameras around;
https://github.com/map7/motioneye_chpos
Backup before using the script please.

closed 2026-10-06 by Marijn0