motioneye-project/motioneye #178: feature-request: show video in html5-browser - expand "run a command" on filesave

ozett, 2016-04-15:
i discovered that .webm files run directly in modern browsers and i access my motioncaptuerd files via a filedirectoy in the browser.

the "run a command" for saving files has no variable for the filename, it would be fantastic to convert the captured file to a .webm format.

it is easy in debian like
`ffmpeg -i $f /var/www/html/motion/$i-$DAY.webm`
[http://rachaels-ubuntu.blogspot.de/2015/12/script-to-convert-motion-captured-avis.html](url)

`avconv -i tears_of_steel_720p.mkv -c:v:0 libvpx -crf 10 -vb 4M -c:a copy   tears_of_steel_vp8.mkv`
[https://wiki.debian.org/ffmpeg#Encode_video_in_VP8](url)

tintamarre, 2016-08-17:
I am using `run-a-command` to send a message to slack (instead of an email). A variable for the filename in the command would be useful. Thx.

ccrisan, 2016-08-17:
Use %f.

tintamarre, 2016-08-17:
Thank you for your quick answer.

I tried several commands (i.e. `echo %f >> /tmp/test.tmp`) but can't make it work ... Do you have an example ?

ccrisan, 2016-08-17:
Hmm, can you check your thread-*.conf file and see what the line starting with `on_event_start` looks like? It should be made up of commands separated by `;`. Depending on what you have chose to happen at the moment of a motion event, you may have multiple commands. One of them should be the command that you have typed in the *Run A Command* field. What does it look like?

tintamarre, 2016-08-17:
`motioneye # cat thread-*.conf | grep on_event_start`

`on_event_start /usr/lib/python2.7/site-packages/motioneye/scripts/relayevent.sh "/etc/motioneye/motioneye.conf" start %t; echo %f >> /tmp/test.tmp`

ccrisan, 2016-08-17:
Ah, scratch that (well partially). I hope you were referring to the *File Storage > Run A Command* option (rather than the one under *Motion Notifications*). The first one should support the `%f` argument and should be found next to the `on_movie_end` and/or `on_picture_save` config options.

tintamarre, 2016-08-17:
[image]

ccrisan, 2016-08-17:
Just use the *File Storage > Run A Command* option.

tintamarre, 2016-08-17:
Better ! 
`duplo motioneye # tail -f /tmp/test.tmp
/var/lib/motioneye/2016-08-17/2016-08-17_09-44-33.avi`

Thx.

ozett, 2016-08-17:
i tested the conversion. the files got suprisingly small. :+1: 
wow. but as i laid them manually into the recordings-direcotry. the dont show up in the motion-eye recordings-gui.

is there a filter? can you add .webm files to show up?

i wanted to test, if i see them running in the recordings preview windows. i am using chrome.
i would be fantastic, if i had the small video in .webm format running in the preview of motioneye-rcordings. ....
:+1:

ccrisan, 2016-08-17:
Webm is not among the supported motion codecs, so motionEye won't recognize them either. As soon as motion starts supporting webm, I'll make sure to add support for them in motionEye as well.

ozett, 2016-08-17:
ok. thanks for reply.
my simple assumption was, that is "would" be displayed in any html-page, as long as the brower supports this.

so i wished myself into the dream, that all files in the recordings dir would be "automagically" played in the preview...as long as the fileformat like .webm could be simply displayed in the right html-browser...

if this is actually not the case, i will use file-hook to convert after recording and view them in a directory where the html-server shows them in a listing..

a pity to jump off that wonderful motioneye-gui for this :cry:

Harvie, 2017-09-29:
Currently there are several file formats supported in modern browsers like chrome. Motion supports swf, flv and mp4. I succesfully played files from motion in chrome. This should be easy to do. At least the basic functionality.

ccrisan, 2018-06-20:
Closing as the functionality is now present in the latest version.

closed 2018-06-20 by ccrisan