motioneye-project/motioneye #179: Save a cropped image of the selected motion area !

michaelanis14, 2016-04-15:
Dear Ccrisan,
This could be a very useful feature, specially of have the cam on, capturing a live street!
Saving only the selected changed area could lead to less file size.

Regards

michaelanis14, 2016-04-17:
very interesting solution : https://www.raspberrypi.org/forums/viewtopic.php?f=43&t=52855

ozett, 2016-04-18:
cool
+1

JamBalaya56562, 2026-11-15:
motion doesn't crop to the motion area, but it passes the area to commands: `%i` and `%J` are its width and height, `%K` and `%L` its centre. So "Run A Command" under File Storage, e.g. `/usr/local/bin/crop-motion %f %i %J %K %L` with a script that runs ImageMagick's `convert "$1" -crop "${2}x${3}+$(($4-$2/2))+$(($5-$3/2))" "${1%.jpg}-crop.jpg"`, saves the cropped part next to each picture. Closing as not planned.

closed 2026-11-15 by JamBalaya56562