motioneye-project/motioneye #180: Is there a way to run motionEye on a windows machine

chfakht, 2016-04-17:
Hi i want to install motionEye on windows , i don't know if it will be possible to use cygwin or does anybody have done this before

thanks

bovine, 2016-04-20:
It's very Linux specific right now. It's be easier to just install a virtual machine with the Linux distribution of your choice.

Alternatively, just get a Raspberry Pi and install the motioneyeos version on it. https://github.com/ccrisan/motioneyeos

sambul13, 2016-04-25:
Its possible to access Motioneye server website on a Windows PC, but for some reason most controls aren't available in a browser in Windows. How to make these controls available on Lan?

bovine, 2016-04-26:
@sambul13 The web interface works fine in Windows. Download and use Chrome, if you think your browser isn't displaying the pages properly. Also be aware that most of the configuration options are not displayed until you select the "Advanced Settings" slider under motionEye's General Settings.

sambul13, 2016-04-26:
How did you configure motioneye to make all its settings accessible on another PC on LAN, including Windows PC? What exact setting controls that?

bovine, 2016-04-26:
If you can load its webpage, then it is accessible. If you are trying to access it from the external side of a router then you may need to do port mappings first, but that is not the fault of motionEye. Regardless, you do need to authenticate regardless of where you access it from.

sambul13, 2016-04-26:
In Linux I can access [cam settings](http://s32.postimg.org/or0sn2abp/motioneye_lin.jpg) in Motioneye webUI. In Windows there're [no settings](http://s32.postimg.org/3k91mbikl/motioneye_win.jpg) accessible in the same webUI. If you can access cam settings in Windows, pls post relevant motioneye config settings (not words).

chfakht, 2016-04-27:
@bovine yes i have tested it on raspberry pi and it's working well but also on ubuntu 14.04 wich works great , However using vmware with ubuntu i haven't been able to configure correctly the laptop webcam and unfortunaltely it's not detected ... i just want to be able to run it on a windows machine : please can you tell mewhere can i find   **The web interface  for Windows** thanks

sambul13, 2016-04-27:
Motion is Linux program, so any interface to motion will work only on Linux. Some web interfaces may allow config via Windows browser, but motioneye is about the only one exposing most cam settings. Still you must have motioneye running on Linux PC or Raspberry to access its webUI from a Windows PC. There might be certain motioneye option that exposes all its settings on LAN.

As well, if your laptop cam works in Ubuntu (with motion?), just take settings from motion.conf and copy them to thread_N.conf in motioneye dir.

bovine, 2016-04-27:
If you intend to run motionEye in a Linux VM on a Windows host, then you'll either need to have the camera be an IP Camera accessed over the network, or use a USB camera on the Host that you configure the VM to have access to.

There is no separate "web interface for Windows", since you can just run the motionEye server on Linux and access its web interface from any Windows web browser. (Running the motionEye server on Windows is not supported. If you are trying to run the motionEye server on Windows using python.exe then that is your problem.)

@sambul13 Even if you haven't configured any authentication, you must still login first using "admin" and blank password to access all of the configuration settings. This has nothing to do with using a Windows web browser. Just close the "hamburger" menu button and click the "person" icon to login. I've seen this work on Chrome, MS Edge, and MSIE on Windows.
[image]

sambul13, 2016-04-27:
Thanks. Login worked. Still I wonder if you know that Motioneye has config files, and a range of configuration settings not accessible from webui? Unless you configure at least 2 options, its not possible to control cams on Lan. :)

```
listen 0.0.0.0
motion_control_localhost false
```
However, if you feel confident in Motioneye, may I invite you to [this thread](https://github.com/ccrisan/motioneye/issues/170)? We are struggling to get motion detection work, and no luck. May be you can help? But pls without "just works" attitude, because... it does NOT. :)

If you still feel motion detection works for you, let us know how your settings are different from ours?

closed 2016-07-03 by ccrisan