motioneye-project/motioneye #181: RTC DS

viejorox, 2016-04-17:
Hi

I have MotionEye running in a Rpi Model B in a remote place whithout network connection, so when power fails, MotionEye restarts with 01/01/1970 date.

Is there anyway to make RTC DS3231 works with MotionEye?

Thanks for your excellent work!

sgiannulli, 2016-04-17:
If you're running Raspbian Jessie try this [https://www.raspberrypi.org/forums/viewtopic.php?f=66&t=125003&p=838314&hilit=ds3231+jessie#p838314](url)
perfectly working for me with this RTC module [http://www.ebay.it/itm/DS3231-Precision-RTC-Clock-Module-Memory-Module-for-Arduino-for-Raspberry-Pi-LS-/162036834565?hash=item25ba25e105:g:VB8AAOSwJMhXDcXV](url)

viejorox, 2016-04-17:
Thank you for your answer, sgiannulli. 

I checked DS3231 and It works fine on Jessie and Wheezy, but I don't know if MotionEye "as is" will load correctly at boot the module needed for the RTC. 

I have problems with the electric supply and it would be very useful for me. I don't want to use powerbanks :)
.

sgiannulli, 2016-04-17:
if you use motioneye on top of raspbian of course it works ( mine does ) as it's a kernel job; if you're talking about motioneyeOS you need Calin to answer

viejorox, 2016-04-17:
I prefer MotionEyeOS :)

Thanks again

sgiannulli, 2016-04-17:
[https://github.com/ccrisan/motioneyeos/issues/61](url)this is not the correct github repo if you're using motioneyeOS
have a look here, maybe it helps

closed 2016-04-18 by viejorox