motioneye-project/motioneye #184: feature-request: bulk delete

ozett, 2016-04-19:
a checkbox or something else to have bulk-deletion of a lots of recordings would be helpful...
:blush:

ozett, 2016-04-21:
i found the delete-all button on the bottom of my recordings page. 
the page was soooo long,... did not see it instantly
assumed the button on top right  :blush:

sambul13, 2016-04-25:
Did you mean the button is placed in a folder, and you see it when the folder is open in File Manager?

ozett, 2016-04-26:
i did not found the button on bottom, as my window was too low down.
maybe one finds the button better, if it is placed more usual on up right...
[image]
[image]

sambul13, 2016-04-26:
Sorry. I was thinking about File Explorer. In Motioneye UI I see it now OK> :)

yanoschik, 2016-07-21:
I'd like to have the check-boxes so I can selectively delete files...

JamBalaya56562, 2026-11-09:
The "Delete all" button you found covers the first part. Deleting selected files only is #232, which stays open, so closing this one as a duplicate of it.

closed 2026-11-09 by JamBalaya56562