motioneye-project/motioneye #185: Archlinux on Raspberry 3 : Cannot install MotionEye 0.31.4

LB72, 2016-04-19:
Hello,

I try to install MotionEye on my Raspberry 3 (updated ArchLinux). I got this error ...

pip2 install --upgrade motioneye
Collecting motioneye
  Using cached motioneye-0.31.4.tar.gz
    Complete output from command python setup.py egg_info:
    /usr/lib/python2.7/distutils/dist.py:267: UserWarning: Unknown distribution option: **'entry_points'**
      warnings.warn(msg)
    /usr/lib/python2.7/distutils/dist.py:267: UserWarning: Unknown distribution option: **'install_requires'**
      warnings.warn(msg)
    usage: -c [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
       or: -c --help [cmd1 cmd2 ...]
       or: -c --help-commands
       or: -c cmd --help

    error: invalid command 'egg_info

Python2, setuptools, MotionEye dependencies are update.

Any idea ?
Thanks

ccrisan, 2016-04-19:
http://stackoverflow.com/questions/12767023/python-packaging

Basically, you need to install setuptools instead of distutils.

LB72, 2016-04-19:
Hi,
setuptools for python2 is already install.

`Packages (1) python2-setuptools-1:20.9.0-1

Total Installed Size:  1.34 MiB
Net Upgrade Size:      0.00 MiB

:: Proceed with installation? [Y/n] y
(1/1) checking keys in keyring                                                                                  [##################################################################] 100%
(1/1) checking package integrity                                                                                [##################################################################] 100%
(1/1) loading package files                                                                                     [##################################################################] 100%
(1/1) checking for file conflicts                                                                               [##################################################################] 100%
(1/1) checking available disk space                                                                             [##################################################################] 100%
:: Processing package changes...
(1/1) **reinstalling python2-setuptools     **                                                                      [##################################################################] 100%`

LB72, 2016-04-19:
Try this :

pip2 install --force-reinstall --upgrade setuptools

and ... It works.

Thank you

closed 2016-04-19 by LB72