motioneye-project/motioneye #187: Install MotionEye on Raspbian Wheezy for RTSP camera

samfin13, 2016-04-20:
Hello,

(I'm a NOOB in linux world :) ) I tried to install MotionEye on my Raspbian Wheezy (https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian with adaptation for Wheezy).
At the end, the web interface is OK, but I can't add my network camera : protocol rtsp not supported.

I did some research and I think the problem is motion.

I have installed ffmeg with apt-get :

```
root@raspberrypi:~# ffmpeg -version
ffmpeg version 0.8.17-6:0.8.17-2+rpi1, Copyright (c) 2000-2014 the Libav developers
   built on Mar 18 2016 02:11:55 with gcc 4.6.3
The ffmpeg program is only provided for script compatibility and will be removed
in a future release. It has been deprecated in the Libav project to allow for
incompatible command line syntax improvements in its replacement called avconv
(see Changelog for details). Please use avconv instead.
ffmpeg 0.8.17-6:0.8.17-2+rpi1
libavutil    51. 22. 3 / 51. 22. 3
libavcodec   53. 35. 0 / 53. 35. 0
libavformat  53. 21. 1 / 53. 21. 1
libavdevice  53.  2. 0 / 53.  2. 0
libavfilter   2. 15. 0 /  2. 15. 0
libswscale    2.  1. 0 /  2.  1. 1
libpostproc  52.  0. 0 / 52.  0.100

```

I have installed "Mr Dave" motion, as recommended for rtsp support and I think that the problem is here : 
```
root@raspberrypi:~# motion -h
motion: error while loading shared libraries: libavutil.so.54: cannot open shared object file: No such file or directory

```

```
root@raspberrypi:/usr/local/bin# ldd motion
        linux-vdso.so.1 =>  (0x7ef4b000)
        /usr/lib/arm-linux-gnueabihf/libcofi_rpi.so (0x76f34000)
        libm.so.6 => /lib/arm-linux-gnueabihf/libm.so.6 (0x76eb2000)
        libpthread.so.0 => /lib/arm-linux-gnueabihf/libpthread.so.0 (0x76e93000)
        libjpeg.so.8 => /usr/lib/arm-linux-gnueabihf/libjpeg.so.8 (0x76e58000)
        libavutil.so.54 => not found
        libavformat.so.56 => not found
        libavcodec.so.56 => not found
        libswscale.so.3 => not found
        libc.so.6 => /lib/arm-linux-gnueabihf/libc.so.6 (0x76d23000)
        /lib/ld-linux-armhf.so.3 (0x54adb000)
```

Some librairies are missing. I tried to install these librairies but I don't know how.

```
root@raspberrypi:/usr/local/bin# apt-get install libavutil54 libavformat56 libswscale3 libavcodec56
Lecture des listes de paquets... Fait
Construction de l'arbre des dÃ©pendances
Lecture des informations d'Ã©tat... Fait
E: Impossible de trouver le paquet libavutil54
E: Impossible de trouver le paquet libavformat56
E: Impossible de trouver le paquet libswscale3
E: Impossible de trouver le paquet libavcodec56
```

(error : packages not found)

Have you any idea ?

blackdot, 2016-04-20:
I have almost same setup as you. Only I am using Jessie.
It looks like you don't have properly configured rapsbian repositories.
For example one of the packages:

>$ apt-cache policy libavcodec56
libavcodec56:
  Installed: 6:11.6-1~deb8u1
  Candidate: 6:11.6-1~deb8u1
  Version table:
 *** 6:11.6-1~deb8u1 0
        500 http://ftp.debian.org/debian/ jessie/main armhf Packages
        500 http://security.debian.org/ jessie/updates/main armhf Packages
        100 /var/lib/dpkg/status<

Can you try the same?
What do you have in `/etc/apt/sources.list`

samfin13, 2016-04-20:
Thanks.
 
The first command is a failure : 
```
root@raspberrypi:~# apt-cache policy libavcodec56
N: Impossible de trouver le paquet libavcodec56
```
(Unable to found the package libavcodec56)

I haven't changed manually the `sources.list` :

```
root@raspberrypi:~# vi /etc/apt/sources.list
deb http://mirrordirector.raspbian.org/raspbian/ wheezy main contrib non-free rpi

# Uncomment line below then 'apt-get update' to enable 'apt-get source'
#deb-src http://archive.raspbian.org/raspbian/ wheezy main contrib non-free rpi
```

ozett, 2016-04-21:
try 

`apt-cache search libav*`

blackdot, 2016-04-21:
Looks like in rapsbian whezzy are different versions of packages:
libavutil51
libavformat53
libswscale2
libavcodec53
These packages should be installable, but not sure if they will satisfy the motion requirements.
I would try to create symlinks from libavutil.so.51 -> libavutil.so.54

How did you install motion? Was it some pre-compiled binary? If yes, it looks like some one compile it on rapsbian jessie.

samfin13, 2016-04-21:
> try
> 
> apt-cache search libav*

The available versions are nearly the same as blackdot says.
libavutil52
libavformat54
libswscale2
libavcodec54

I tried the symlinks : 
```
lrwxrwxrwx 1 root root 19 avril 21 16:57 libavutil.so.54 -> libavutil.so.52.3.0
lrwxrwxrwx 1 root root 17 avril 21 17:00 libavformat.so.56 -> libavformat.so.54
lrwxrwxrwx 1 root root 15 avril 21 17:02 libswscale.so.3 -> libswscale.so.2
lrwxrwxrwx 1 root root 16 avril 21 17:01 libavcodec.so.56 -> libavcodec.so.54
```

but it doesn't work : 
```
root@raspberrypi:/usr/lib/arm-linux-gnueabihf# motion -h
motion: /usr/lib/arm-linux-gnueabihf/libswscale.so.3: version `LIBSWSCALE_3' not found (required by motion)
motion: /usr/lib/arm-linux-gnueabihf/libavutil.so.54: version `LIBAVUTIL_54' not found (required by motion)
motion: /usr/lib/arm-linux-gnueabihf/libavformat.so.56: version `LIBAVFORMAT_56' not found (required by motion)
motion: /usr/lib/arm-linux-gnueabihf/libavcodec.so.56: version `LIBAVCODEC_56' not found (required by motion)
```

I installed the precompiled Mr Dave version supplied by ccrisan (https://github.com/ccrisan/motioneye/wiki/precompiled/motion-mrdave-raspbian), certainly a version for Jessie.

Do I have to install another version of motion ?

samfin13, 2016-04-21:
I found this : 
```
root@raspberrypi:/# find / | grep -E "libavutil.so.54|libavformat.so.56|libavcodec.so.56|libswscale.so.3"
/usr/lib/omxplayer/libavformat.so.56
/usr/lib/omxplayer/libavcodec.so.56
/usr/lib/omxplayer/libswscale.so.3
/usr/lib/omxplayer/libavcodec.so.56.26.100
/usr/lib/omxplayer/libavutil.so.54.20.100
/usr/lib/omxplayer/libswscale.so.3.1.101
/usr/lib/omxplayer/libavutil.so.54
/usr/lib/omxplayer/libavformat.so.56.25.101
```
Maybe, I can try to use this files for motion … but I don't know how ( path ?)

blackdot, 2016-04-21:
uninstall the libav packages and then create new symlinks... example:
ln -s /usr/lib/omxplayer/libavcodec.so.56 /usr/lib/arm-linux-gnueabihf/libavutil.so.56

samfin13, 2016-04-21:
I have uninstalled the 4 libav packages
I have created  symliks for this 4 packages (and another one more for libswresample.so.1).
```
root@raspberrypi:~# motion -h
motion Version Unofficial-Git-UNKNOWN, Copyright 2000-2005 Jeroen Vreeken/Folkert van Heusden/Kenneth Lavrsen

usage:  motion [options]

Possible options:

-b                      Run in background (daemon) mode.
-n                      Run in non-daemon mode.
-s                      Run in setup mode.
-c config               Full path and filename of config file.
-d level                Log level (1-9) (EMG, ALR, CRT, ERR, WRN, NTC, INF, DBG, ALL). default: 6 / NTC.
-k type                 Type of log (COR, STR, ENC, NET, DBL, EVT, TRK, VID, ALL). default: ALL.
-p process_id_file      Full path and filename of process id file (pid file).
-l log file             Full path and filename of log file.
-m                      Disable motion detection at startup.
-h                      Show this screen.

Motion is configured using a config file only. If none is supplied,
it will read motion.conf from current directory, ~/.motion or /etc/motion.
```
:)

In motionEye web interface, I tried to add my camera : "rtsp protocol not supported" :(.
I restrart my pi.
In motionEye web interface, I have added my camera with succes :).

Thanks :)

closed 2016-04-21 by samfin13