motioneye-project/motioneye #189: No longer uploads to Google Drive

delize, 2016-04-22:
Hey,

Initially, I had Google Drive setup, and everything was working alright. Recently, I moved houses, and setup the motioneye interface and camera again, and after turning everything on, and expecting photos uploaded to Google Drive, it failed to work.

Nothing has changed (except for the Public facing IP) in terms of connectivity, Google Drive is still accessible, and config wise, it is still configured to properly use the Google account with Oauth2, but it just doesn't seem to recognize the upload. I have tried twice using a different key.

The only info I can pull from /var/log/motion.log is this:

> 2016-04-20 15:07:08: [sendmail]     INFO: email sent
> 2016-04-20 15:16:35: [sendmail]     INFO: email sent
> 2016-04-20 15:16:53: [sendmail]     INFO: email sent
> 2016-04-21 07:41:47: [sendmail]     INFO: email sent
> 2016-04-21 07:42:16: [sendmail]     INFO: email sent
> 2016-04-21 07:42:43: [sendmail]     INFO: email sent
> 2016-04-21 07:43:03: [sendmail]     INFO: email sent
> 2016-04-21 07:43:48: [sendmail]     INFO: email sent
> 2016-04-21 07:44:12: [sendmail]     INFO: email sent
> 2016-04-21 07:44:54: [sendmail]     INFO: email sent
> 2016-04-21 07:46:01: [sendmail]     INFO: email sent
> 2016-04-21 07:48:32: [sendmail]     INFO: email sent
> 2016-04-21 07:49:42: [sendmail]     INFO: email sent
> 2016-04-21 07:49:49: [sendmail]     INFO: email sent
> 2016-04-21 07:50:04: [sendmail]     INFO: email sent
> 2016-04-21 07:50:24: [sendmail]     INFO: email sent
> 2016-04-21 17:56:35: [sendmail]     INFO: email sent
> 2016-04-21 17:57:19: [sendmail]     INFO: email sent
> 2016-04-21 18:05:32: [sendmail]     INFO: email sent
> 2016-04-21 18:06:48: [sendmail]     INFO: email sent
> 2016-04-21 18:07:00: [sendmail]     INFO: email sent
> 2016-04-21 18:07:23: [sendmail]     INFO: email sent
> 2016-04-21 18:07:51: [sendmail]     INFO: email sent
> 2016-04-21 18:11:25: [sendmail]     INFO: email sent

I don't have any more helpful info. If I try to use the online test tool then I get the following information:
[image]

So I am not sure what is going on.

ccrisan, 2016-04-22:
I'd need to see the motioneye.log file. If you're under systemd, you can get that using journactl -u motioneye > motioneye.log.

Ethocreeper, 2016-04-22:
i have the same problem
my [issue ](https://github.com/ccrisan/motioneye/issues/188)is older (2 days) haven't answered an this 12 hours.

delize, 2016-04-23:
@ccrisan Hey, I am unfortunately, not on journalctl.

> $ journalctl -u motioneye > motioneye.log
> No journal files were found.

@Ethocreeper Your problem doesn't appear to be related to mine at all.

sgiannulli, 2016-04-24:
@delize
You need sudo before the comand

sambul13, 2016-04-25:
@delize 
try to find the logs:
sudo find / -name "motioneye.*"
which motioneye

chaseacton, 2016-06-27:
Having the same problem. I have two cameras, both uploading to Drive. Randomly one of them stopped uploading. I didn't change anything. I've generated new keys with no luck.

ccrisan, 2016-06-29:
@chaseacton I'd need to see your motioneye log file. Setting the log level to debug would help even further.

ccrisan, 2016-07-02:
I have made some changes to the google drive uploader. Tomorrow I'm hoping to release 0.32.1. Please try it when you can and see if the problem still persists.

chaseacton, 2016-07-02:
@ccrisan How can I change the log level?

ccrisan, 2016-07-02:
@chaseacton, see https://github.com/ccrisan/motioneye/wiki/Configuration-File#log_level

chaseacton, 2016-07-02:
Thanks. If the next update doesn't fix Drive uploading, I'll upload my logs.

ccrisan, 2016-07-02:
Please do. The update is already available, by the way.

delize, 2016-07-03:
I attempted to use the update but i am getting:
Accessing the upload service failed: invalid_grant!

ccrisan, 2016-07-03:
@chaseacton Judging from the fact that we're on motionEye's issue tracker, I thought you were using motionEye, not motionEyeOS. The new motionEyeOS version is not ready yet.

ccrisan, 2016-07-03:
@delize have you tried to repeat the authorization process from the beginning?

delize, 2016-07-03:
Hey,

I assume you mean, attempting to generate a new key and entering it? 
Yes, I did that. I did not however, revoke the previous application assignment in my Google Account if that was required.

ccrisan, 2016-07-03:
Revoking the old one should never be necessary (I wouldn't know how to do that, btw). At what point do you get "invalid_grant"? Could you please try removing the /etc/motioneye/uploadservices.json file and try again?

delize, 2016-07-04:
After deleting that file, I now get the following error when trying to generate a key:
{"error": "unknown upload service gdrive"}

ccrisan, 2016-07-04:
How about enabling and then disabling upload services (hitting apply each time)?

delize, 2016-07-13:
This is no longer a problem, but, not for a reason that it was magically fixed. I inadvertently broke my MicroSD card, and had to start from scratch. Currently uploads to Google Drive work well. :( So I can't be of much more help unless the problem re-appears in a bit.

jeff-h, 2016-09-16:
I've never had gdrive set up with motionEye, but previously (with an earlier version of motionEye) had dropbox working with no problems.

Tonight I tried to set up Google Drive and everything appears configured properly, my `uploadservices.json` looks fine, no errors (that I can see) in the log. I updated to v0.33.4 and still no improvement.

Any thoughts?

ccrisan, 2016-09-16:
Hmm this upload services feature is killing me. @jeff-h could you please set the log level to debug and attach the resulted log here?

However if you cannot see any error in your log, then the root cause could also be a configuration-related issue. Here are some helping questions:

 * Have you configured more than one camera on your setup?
 * Have you enabled/disabled cameras?
 * Are media files actually created locally on your disk?
 * Have you enabled the uploading of the particular media type (i.e. movies/pictures) you're interested in?

chaseacton, 2016-09-16:
Having the exact same problem. One of my cameras will not upload to G drive. I can see the images captured on the SD card at /data/output. Both cameras have identical settings. I just switched my log level to debug and will report back after triggering motion.

jeff-h, 2016-09-17:
@ccrisan Thanks for putting in so much effort to the issue queue, it's really appreciated. 

It occurred to me I'd have no problems opening up SSH access to my box for you if you wanted to have a play with a live install exhibiting the problem. The biggest problem with that would probably be that it is running motionEye inside a Docker container which means there's a bit of a layer in the way.

Anyway, I have set my log level to debug but was very surprised to see that my `motioneye.conf` was completely empty. I am also not getting anything at all in the log—i.e. the log is zero bytes, so I'm thinking I have some side issues to sort out first :(

jeff-h, 2016-09-17:
Sorry, I forgot to answer your questions.

* I have 3 cams, and initially set up all three with google drive. When that failed, I cleared out `uploadservices.json` entirely and started again, this time only setting up the first camera with google drive.
* I'm not certain what you mean sorry — all my cameras are active
* the media files (pictures and videos) work beautifully from all 3 cameras, and appear on disk and in the motionEye UI as expected
* in motionEye, I have set `Upload Media Files`, `Upload Pictures` and `Upload Movies` to `ON`.

ccrisan, 2016-09-17:
Well remote access would indeed be the best thing I could hope for in this case. If you're kind enough to grant me SSH access, that would be great.

chaseacton, 2016-09-19:
I'm seeing a lot of these where I believe uploads to gdrive should be taking place: WARNING: 403 POST /_relay_event/?_username=myusername&event=start&thread_id=1&_signature=04fef025e737dfa155327328aed19f7fd7fdf701 (127.0.0.1) 101.01ms

ccrisan, 2016-09-19:
@chaseacton yeap that's an authentication issue. Do you have any special characters in your admin password? Is this the latest version?

chaseacton, 2016-09-19:
Wow, that was it for me! Tons of special characters in my password. As soon as I changed it, the upload service started working again. Is there a fix planned for this?

ccrisan, 2016-09-19:
Well could you please tell me what were those special characters (in random order)? I'll need to see which of them needs an extra escaping.

chaseacton, 2016-09-19:
&^(%=(]?#>

It was also a 50 character password in case length could be a factor here.

ccrisan, 2016-10-26:
0d4e7dc01221747769f762667e91448a25f6e0da fixes this.

closed 2016-10-26 by ccrisan