motioneye-project/motioneye #190: feature-request: overlay image for active motion-detection

ozett, 2016-04-25:
as i investigated another issue,
i found the http-api. where one could check actual status of motion, and motion-detection on itself.

`http://192.168.14.212:7999/1/detection/status`

may you think over creating an overlay image for recording, means motion-dections active?

sambul13, 2016-04-26:
Are you talking of cam frame changing color to red? Its already implemented, and works when motion is recording to disk.

ozett, 2016-04-27:
somethin like this, i thought...
[image]

sambul13, 2016-04-27:
It would be good if the cam frame remains black despite detection is enabled, so user can't figure out what went wrong, and why there are no saved filed on video and image folders. However, we still need to find the root cause and how to fix this problem.