motioneye-project/motioneye #191: MotionEye OS

GitYaSome, 2016-04-25:
Hi there - I have the MotionEye OS installed on my Pi3 and everything is working great with my built in camera.  

Today I got the 7" touchscreen in the mail, added it to the Pi3 and need to run the following 3 commands:
sudo apt-get update
sudo apt-get upgrade
sudo apt-get install matchbox-keyboard

How do I kill MotionEye and get to a command prompt?

I am sitting at this prompt:
meye-28ae4616 login:

I can login as admin / no password and get to:
[root@meye-28ae4616 ~]# 

But don't know how to stop MotionEye and just get to the OS.

Thanks for any pointers.

sambul13, 2016-04-25:
This is not a valid "issue" with the package. Just read the Wiki or ask the author to setup Motioneye forum, where users can help each other and exchange information. Its especially important since there is no normal Motion forum. Not only its offered in obsolete and hard to use format, but because of that no-one ever answers on that forum for years. So Motioneye forum would definitely be handy.

GitYaSome, 2016-04-26:
Thanks - I hope that can be made.
In the mean time, if anyone has any ideas - Please pass them my way...

ccrisan, 2016-04-26:
@GitYaSome, see this https://github.com/ccrisan/motioneyeos/wiki/FAQ#i-cant-seem-to-find-apt-get-or-any-other-package-manager-what-gives.

closed 2016-04-26 by ccrisan