motioneye-project/motioneye #193: Motioneye can't control motion detection

sambul13, 2016-04-26:
Once motioneye service is started, detected motion pictures and videos aren't saved into target directory or anywhere. There're no relevant records visible in motioneye or motion log. It appears, setting _control_html_output_ on or off isn't recognized by Motioneye. 

Only if webcontrol_html_output is turned on, its possible to activate detected motion recording to disk via native Motion webui. But as soon as any changes are made in Motioneye webui, motion detection is set to Pause again by Motioneye, despite its shown active in its webui. But cam picture is not red framed.

sambul13, 2016-04-26:
Here's this [pic](http://s32.postimg.org/vzw1ittxh/motioneye_detection.jpg) showing motioneye puts detection to PAUSE at each settings change of any cam.

ccrisan, 2016-07-03:
Most likely you have the working schedule option turned on.

closed 2016-07-03 by ccrisan