motioneye-project/motioneye #195: Feature request: wait for partition to be mounted on boot

influenced, 2016-05-01:
On my Pi w/ Raspbian, I have an external USB drive which at boot doesn't get mounted till 5-10 seconds after the MotionEye service is started - that is to say, in the camera's config under Storage Device I have Toshiba External (/dev/sda1) and a relative path under Root Directory. This causes MotionEye to fail on startup at boot time, since it tries to look up the base mount path for that device and it doesn't yet exist. If I remove the device dependency and just specify a custom absolute path (/media/pi/DATA/motioneye/xxx or similar), I have a different problem - motioneye auto-creates those directories and the drive gets forcibly mounted to a different mountpoint (/media/pi/DATA1), so instead of not launching motioneye ends up writing to the SD card instead of the drive.

While I can make motioneye.service depend on that mount* with systemd magic, a more usable solution for the less technical might be to have a global setting or camera-specific checkbox which makes motioneye wait (perhaps with an upper time limit) for that device's mountpoint to exist in config.py.

* in fact, being a systemd novice, I never got that to work - instead I lazily made motioneye.service point to a script which just checks if the device exists and has been mounted before calling meyectl, sleeping for 5 seconds and trying again if it hasn't. Note that the block-level device (/dev/sda1) itself shows up when the service is launched, it just doesn't get mounted (/media/pi/DATA) till after.

maglub, 2016-07-03:
The systemd magic would be:

1) Get the name of the systemd component to wait for:

```
systemctl | grep mnt
```

For example it could be called "mnt-mymediadrive.mount" or something else.

2) Put this into /etc/systemd/system/motioneye.service (the first two lines are already there).

```
[unit]
Description=motionEye Server
After=mnt-mymediadrive.mount
...
```

Then reboot and see that your drive is mounted before motioneye was started.

Putting this kind of dependency into the application will add a complexity to the startup mechanism that is taken care of in systemd. If it were to be put into the application, I would suggest that the application should check the /etc/systemd/system/motioneye.service file for this entry and handle it there.

influenced, 2016-07-03:
That's all correct, and I did get that to work in the end. I just feel that the current motioneye behaviours with an unmounted drive - outright failure, or replicating the directory structure forcing the mountpoint to change - are undesirable.

maglub, 2016-07-04:
But then, how would motioneye "know" that you intend to mount a drive onto that directory?

I would say that it works as designed (not that I have anything with the design to do).

influenced, 2016-07-04:
In fact, it already does something very close to what is needed - it checks to see the device exists (which it does) and incorrectly assumes that means it has been mounted (which it hasn't yet been); the failure occurs when it grabs the mountpoint path, which is null.

shermanlai, 2017-06-03:
I wasn't so lucky to get my drive mounted before motioneye starts.  When I `systemctl` I don't get any mnt services, but instead got

`media-pi-64GBUSB4.mount   loaded active mounted   /media/pi/64GBUSB`

not sure why I don't get a mnt-mymediadrive.mount. 

I've tried all variations of in `/etc/systemd/system/motioneye.service`
`RequiresMountsFor=media-pi-64GBUSB4.mount`
-OR-
`After=/dev/sda1`

JamBalaya56562, 2026-11-09:
As maglub said, systemd is the right place for this: add `RequiresMountsFor=/media/pi/DATA` (the mount path, not the unit name) to the `[Unit]` section of motioneye.service, with `systemctl edit motioneye`, and motionEye starts only after the drive is mounted. Closing, since this needs no setting in motionEye.

closed 2026-11-09 by JamBalaya56562