motioneye-project/motioneye #197: FFMPEG vs. LIBAV

asprunt, 2016-05-06:
Your installation instructions describe the process for using Motion with FFMPEG, but not with LIBAV. The [compile instructions](https://github.com/Mr-Dave/motion/blob/master/INSTALL) for Mr. Dave's fork describe both. LIBAV would seem to be more convenient since it's available in package form on debian & raspbian. Is there a reason for this? Is LIBAV inferior in some way?

ccrisan, 2016-05-06:
Yes, there is. Even though they say libav is somehow compatible with ffmpeg, the ffmpeg command line utility is missing. motionEye makes intense use of ffmpeg for various non-motion-related tasks (e.g. thumbnails or timelapse) and without it those functions wouldn't be available. I'm not saying that it wouldn't work, but not having those features would rise too many questions to which I simply don't have time to answer.

asprunt, 2016-05-06:
Thanks for clearing that up. Does the pre-compiled version of FFMPEG linked to in your instructions include the h.264 encoder and decoder?

ccrisan, 2016-05-06:
# ffmpeg --help
    ffmpeg version 2.7.2 Copyright (c) 2000-2015 the FFmpeg developers
    built with gcc 4.8.5 (Buildroot 2015.08-rc1-g3394b0c)
    configuration: --enable-cross-compile --cross-prefix=/home/ccrisan/motioneyeos/output/raspberrypi/host/usr/bin/arm-buildroot-linux-gnueabihf- --sysroot=/home/ccrisan/motioneyeos/output/raspberrypi/host/usr/arm-buildroot-linux-gnueabihf/sysroot --host-cc='/home/ccrisan/motioneyeos/output/raspberrypi/host/usr/bin/ccache /usr/bin/gcc' --arch=arm --target-os=linux --disable-stripping --pkg-config=/home/ccrisan/motioneyeos/output/raspberrypi/host/usr/bin/pkg-config --disable-static --enable-shared --prefix=/usr --enable-avfilter --disable-version3 --enable-logging --enable-optimizations --disable-extra-warnings --enable-avdevice --enable-avcodec --enable-avformat --disable-x11grab --enable-network --disable-gray --enable-swscale-alpha --disable-small --enable-dct --enable-fft --enable-mdct --enable-rdft --disable-crystalhd --disable-vdpau --disable-dxva2 --enable-runtime-cpudetect --disable-hardcoded-tables --disable-memalign-hack --disable-msa --enable-hwaccels --disable-avisynth --disable-frei0r --disable-libopencore-amrnb --disable-libopencore-amrwb --disable-libopencv --disable-libdc1394 --disable-libfaac --disable-libgsm --disable-libmp3lame --disable-libnut --disable-libopenjpeg --disable-librtmp --disable-libschroedinger --disable-libspeex --disable-libtheora --disable-libvo-aacenc --disable-libvo-amrwbenc --disable-symver --disable-doc --enable-gpl --enable-nonfree --enable-ffmpeg --disable-ffplay --disable-ffserver --disable-avresample --disable-ffprobe --disable-postproc --enable-swscale --enable-indevs --enable-outdevs --enable-pthreads --enable-zlib --disable-bzlib --enable-openssl --disable-vaapi --disable-libopus --disable-libvpx --enable-libfreetype --disable-fontconfig --disable-libx264 --disable-yasm --disable-mmx --disable-sse --disable-sse2 --disable-sse3 --disable-ssse3 --disable-sse4 --disable-sse42 --disable-avx --disable-avx2 --enable-armv6 --enable-vfp --enable-mipsfpu --disable-mips32r2 --disable-mipsdspr1 --disable-mipsdspr2 --disable-altivec --enable-pic --cpu=arm1176jzf-s

ccrisan, 2016-05-06:
That's the one motionEyeOS uses. The one for Raspberry PI to be used with Raspbian might have it (I don't have a Raspbian setup ready right now) but I it's not used by motion and that's for sure.

closed 2016-07-03 by ccrisan