motioneye-project/motioneye #199: using higher H264 streams and non standard resolution with MotionEye

Eisbahn, 2016-05-08:
Hello,

just made my first steps with MotionEye and my camera (consumer 3MP one from Canon). After first steps and struggeling on installation (normally using Windows :=( as OS), I do have more questions, but first what HW/SW I'm using:
- gigabit switched network, all devices connected physically via LAN
- as PC I'm using an older ACER Revo (Intel(R) Atom(TM) CPU  330   @ 1.60GHz) with Ubuntu 14 as my old RPi Model B seems to be quite too slow. Kodi is running in parallel, but not making that much as it's idle during my tests 
`top - 15:54:49 up 21 min,  2 users,  load average: 5.59, 5.32, 3.66
Aufgaben: 130 total,   1 running, 129 sleeping,   0 stopped,   0 zombie
%Cpu(s): 95.3 be,  3.3 sy,  0.2 ni,  0.2 un,  0.0 wa,  0.0 hi,  1.1 si,  0.0 st
KiB Mem:   1544516 total,  1076876 used,   467640 free,    49392 buffers
KiB Swap:  1570812 total,        0 used,  1570812 free.   323196 cached Mem

  PID BENUTZER  PR  NI    VIRT    RES    SHR S  %CPU %MEM     TIME+ COMMAND
 2595 root      20   0  424440 324092   2412 S 386.0 21.0  34:44.12 motion
 1614 xbmc      20   0  644372 261164  40552 S  10.2 16.9  11:36.29 kodi.bin
 2003 root      20   0   57884  14660   2764 S   2.3  0.9   0:56.57 meyectl
 2613 xbmc      20   0    5496   1480   1080 R   0.7  0.1   0:00.45 top`
- ffmpeg was updated manually to `N-79139-gde1a0d4 Copyright (c) 2000-2016 the FFmpeg developers` with x264 (and 265) support
- motion was updated manually, too (fork from Mr. Dave)
- motioneye is the actual one
- camera could deliver an H264 stream or a MJPEG one and is attached via rtsp/ethernet. 

And now the the questions came up...

- How could I make usage of higher resolutions my camera offers? At the moment I could only pick up to 1600x1200 in MotionEye. Sadly I could set my camera just to 1MP smallest for H264 (which I'm using)
- MotionEye allows me to select UDP and TCP. On the UDP stream I do have lots of blocks/noise/corrupted images inside, TCP seems to be better, therefore I picked this one. But from my understanding TCP should be more worse as I do get TCP/IP overhead involved. What's going on?
- if a motion is detected, the Acer REvo seems to lack on CPU power: lots of frames are dropped making the video more or less to just a quicker slideshow. In the live view of MotionEye I get information like (streaming/capture frame rate) ~12/~5 (around twelve and five). What CPU should be present to make motion detection on rtsp stream and save a video to local HDD for high res video?
- Could I work with an own motion config including a private area by pgm file? and would this reduce the load?

Basic question: do I have just a worse setup and want too much from old HW? How are real CCTV cams making the motion detection with poor/limited HW?

Thanks for any help

ccrisan, 2016-05-09:
1. I believe 1600x1200 is a big enough resolution. Anything higher than that would make motion stutter and the performance will suffer. That's the highest resolution motionEye knows about. 
2. From the reliability and "lossiness" point of view, UDP is always worse than TCP, since there is no confirmation of packet arrival or correct packet ordering (sequencing).
3. A more powerful CPU :) it depends on the resolution, framerate and movie quality.
4. You could use your own config options for motion, in addition to motionEye's or completely separated. I don't see why not.
5. In my opinion the motion daemon is far from an efficient solution for motion detection. Everything is done "in software" by the CPU and relies on frame differences rather than on motion vectors. There's nothing I can do to improve this.

Eisbahn, 2016-05-13:
Hello Calin,

thanks s lot for your answer. Think I had complete wrong impression because of missing background and technical knowledge. My intention was not to blame you or get new SW. I've seen a CCTV (was a Logitech one) which does everything in its small camera housing without anything else and tried some steps on my own with given HW.
Goal was to check kids and dog ;=) So lower resolution would be fine for me, but sadly in this case I need to buy new camera (as my only provides this high resolution output). However: this costs "just" 50€ and everything is inside.

Thanks a lot and dood work

ccrisan, 2016-05-15:
I recommend buying an official RPi camera as it's by far the best one and it's about 25 EUR if I recall correctly.

Oxize, 2016-06-12:
Ive bought the Official RPI camera also. Its good for recording for few metres. When you want to record something outside as a security camera, its lacks focus since its fixed and not changable. But it still gives an nice output. It runs here 30fps in stream and 14-28fps from the camera in 1280x720. 

Ive also got an MS Lifecam HD, which has a verry sensitive low-light sensor and works verry well in low light. I want to try this one out soon how it works with MotionEye.

ccrisan, 2016-07-03:
I no longer know what this issue is about.

closed 2016-07-03 by ccrisan