motioneye-project/motioneye #200: Google drive delete files

softmillennium, 2016-05-09:
No way to delete all motion files in google drive folder once it fills up to make room for newer ones ?

chaseacton, 2016-06-27:
I wrote this Google Script to do this. You may need to fine tune it based on how much your cameras are recording and how much Drive space you have. Run this every day or so with a cron (Google calls these time-based triggers).

`/*
motionEye Drive Deleter
--------------------------
motionEye uses Google Drive to store pictures and videos when triggered by motion.
These files quickly fill up Drive so we need a way to remove files older than a given date.
The basic logic to to specify the base motionEye directory, get its subdirectories, search 
for files in those directories older than the specified age, and delete them.
*/

//Config
var minimumAge = 8; //Delete anything older specified number of days
var motionEyeDirectory = "your_directory_id";

function getFilesByDate(){
  //Array of file IDs to delete
  var filesToDelete = [];

  //Set our target date to search for and create a proper Date object
  var minimumDate = new Date().getTime()-3600*1000*24*minimumAge; //Number of days old
  var cutOffDate = new Date(minimumDate);
  var cutOffDateAsString = Utilities.formatDate(cutOffDate, "GMT", "yyyy-MM-dd");    
  
  //Get root camera storage directory
  var securityFolder = DriveApp.getFolderById(motionEyeDirectory);
  
  //Each camera has its own subdirectory within the root camera directory. Get those folder objects
  var cameras = securityFolder.getFolders();
    
  //Iterate through camera subdirectories and search their files
  while (cameras.hasNext()){
    var folder = cameras.next();
    
    //Search for files in directory that are older than the target date
    var directoryContents = folder.searchFiles('modifiedDate < "' + cutOffDateAsString + '"');

    while (directoryContents.hasNext()){
      var file = directoryContents.next();
      //Logger.log("Count: " + file.getName())
      filesToDelete.push(file.getId());
    }
  }

  return filesToDelete;
};

function deleteFiles() {
  var arrayIDs = getFilesByDate();

  for (var i=0; i < arrayIDs.length; i++) {
    //Logger.log('arrayIDs[i]: ' + arrayIDs[i]);
    //This deletes a file without needing to move it to the trash
    var rtrnFromDLET = Drive.Files.remove(arrayIDs[i]);
    Utilities.sleep(100); //Prevent exceeding rate limit (currently 10 requests per second)
  }
};`

ilyalavrenov, 2017-01-04:
I got around this by setting up rclone with my Google drive as a destination and running something like this in a cronjob every night:

`rclone -q --exclude *.thumb sync /var/lib/motioneye gdrive:motioneye`

At first I did try and set up a Google Apps script similar to the above, but I had issues getting it to delete all of the correct files consistently. Sometimes it would leave a few left over. 

This way you can have Motioneye handle the real-time uploading to Google Drive, and then rclone can sync up the changes at midnight.

JamBalaya56562, 2026-11-09:
The Clean Cloud switch in the storage settings (Google Drive only) does this now: when motionEye removes old local media, it also removes the matching folders on Drive. It is in motionEye since 0.40, so closing.

closed 2026-11-09 by JamBalaya56562