motioneye-project/motioneye #201: Remote code execution

Informatic, 2016-05-10:
Hi, please don't use `shell=True`, especially when no argument is properly escaped, as it allows for (hopefuly authenticated, but I haven't really look into it much) remote for code execution.

https://docs.python.org/2/library/subprocess.html#frequently-used-arguments

```
Camera 1 has to be local or remote motion.
http://localhost:8765/movie/1/preview/%22;touch%20/tmp/oops;%22/
```

Affected code for this specific example is in `motioneye.mediafiles.make_movie_preview`, however I can you are using `subprocess.call`/`subprocess.Popen` with `shell=True` and unescaped arguments all over the place.

ccrisan, 2016-05-18:
Commit 9819d2f should fix all the subprocess-related problems. Thanks again for noticing this problem.

closed 2016-05-18 by ccrisan

Informatic, 2016-05-18:
Cool. I've added some comments to your latest commit.