motioneye-project/motioneye #202: Issue installing on Ubuntu 16.04

jlowe42, 2016-05-12:
Hello,
I'm a linux noob, and trying to follow the directions to install MotionEye on my freshly installed Ubuntu machine.
I get to step 5:

5. Prepare the configuration directory:

mkdir -p /etc/motioneye
cp /usr/local/share/motioneye/extra/motioneye.conf.sample /etc/motioneye/motioneye.conf

Of course, making the directory works fine, but the second command - the copy command dies.  There is no /usr/local/share/motioneye folder (or the files and directories I'm assuming need to be there) - and therefore nothing to copy.

The actual message is:
cp: cannot stat '/usr/local/share/motioneye/extra/motioneye.conf.sample': No such file or directory

Did I miss a step somewhere?

Thanks - and sorry if this is a silly question!

ccrisan, 2016-05-12:
It might be `/usr/share/...` instead of `/usr/local/share/...`.

brik2k, 2016-05-12:
I think I ran into a similar issue. I reran the entire install process after running sudo -i and that seemed to do the trick.

jlowe42, 2016-05-13:
Thank you for the answers.  
The files were not in /usr/share/ either (I looked there last night - sorry for not adding that).

Doing sudo -i and redoing the install corrected the problem.

Thanks!

closed 2016-05-13 by jlowe42

2pizzas, 2019-04-11:
I have the same problem, the files are in ~/.local/bin and ~/.local/share
Changing motioneye.service with these dirs doesn't solve the issue.