motioneye-project/motioneye #204: error while installing motion eye

Ethocreeper, 2016-05-14:
**i get these help**

root@raspberrypi:~# pip install motioneye
Downloading/unpacking motioneye
  Downloading motioneye-0.31.4.tar.gz (556kB): 556kB downloaded
  Running setup.py (path:/tmp/pip-build-MtD7OC/motioneye/setup.py) egg_info for package motioneye

Downloading/unpacking tornado>=3.1 (from motioneye)
  Downloading tornado-4.3.tar.gz (450kB): 450kB downloaded
  Running setup.py (path:/tmp/pip-build-MtD7OC/tornado/setup.py) egg_info for package tornado

    no previously-included directories found matching 'docs/build'
    warning: no files found matching 'tornado/test/README'
Downloading/unpacking jinja2 (from motioneye)
  Downloading Jinja2-2.8-py2.py3-none-any.whl (263kB): 263kB downloaded
Requirement already satisfied (use --upgrade to upgrade): pillow in /usr/lib/python2.7/dist-packages (from motioneye)
Downloading/unpacking pycurl (from motioneye)
  Downloading pycurl-7.43.0.tar.gz (182kB): 182kB downloaded
  Running setup.py (path:/tmp/pip-build-MtD7OC/pycurl/setup.py) egg_info for package pycurl
    Traceback (most recent call last):
      File "<string>", line 17, in <module>
      File "/tmp/pip-build-MtD7OC/pycurl/setup.py", line 823, in <module>
        ext = get_extension(sys.argv, split_extension_source=split_extension_source)
      File "/tmp/pip-build-MtD7OC/pycurl/setup.py", line 497, in get_extension
        ext_config = ExtensionConfiguration(argv)
      File "/tmp/pip-build-MtD7OC/pycurl/setup.py", line 71, in __init__
        self.configure()
      File "/tmp/pip-build-MtD7OC/pycurl/setup.py", line 107, in configure_unix
        raise ConfigurationError(msg)
    __main__.ConfigurationError: Could not run curl-config: [Errno 2] No such file or directory
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):

  File "<string>", line 17, in <module>

  File "/tmp/pip-build-MtD7OC/pycurl/setup.py", line 823, in <module>

    ext = get_extension(sys.argv, split_extension_source=split_extension_source)

  File "/tmp/pip-build-MtD7OC/pycurl/setup.py", line 497, in get_extension

    ext_config = ExtensionConfiguration(argv)

  File "/tmp/pip-build-MtD7OC/pycurl/setup.py", line 71, in __init__

    self.configure()

  File "/tmp/pip-build-MtD7OC/pycurl/setup.py", line 107, in configure_unix

    raise ConfigurationError(msg)

__main__.ConfigurationError: Could not run curl-config: [Errno 2] No such file or directory

----------------------------------------
Cleaning up...
Command python setup.py egg_info failed with error code 1 in /tmp/pip-build-MtD7OC/pycurl
Storing debug log for failure in /root/.pip/pip.log

Ethocreeper, 2016-05-18:
anyone?

ccrisan, 2016-05-18:
Well, have you installed all the required dependencies, including `curl` and `libcurl4-openssl-dev`?

Ethocreeper, 2016-05-21:
i foloded all the instructions from [here](https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian)

Ethocreeper, 2016-05-21:
sorry i just fixed it i reformated the sd and then it worked fine so i am closing this issue

Ethocreeper, 2016-05-21:
ok ive just writed the image to the sd card and it worked now i just need username and pass

Ethocreeper, 2016-05-21:
ok with blank pass

Ethocreeper, 2016-05-21:
pls add in donations with paysafe
like you enter how much you wont to donate and then you put you pin on

i am closing this issue beacuse the proble is solved

Ethocreeper, 2016-05-21:
and in putty it asks for pasword of user pi@(ipaddress)

closed 2016-07-03 by ccrisan

tarekstronghol, 2018-10-27:
Please update setup tools _**### pip install -U setuptools**_