motioneye-project/motioneye #205: Motioneye only showing Preferences

AndyR88, 2016-05-14:
hi
I installed the motioneye image file for raspberry pi 3 (latest version).
I can access the cameras howver it does not provide me with a complete settings menu, only "preferences". See attachment.
How can I access the full setting menu?
Thanks!
[image]

AndyR88, 2016-05-15:
I solved it by logging in using username as admin and leaving the password entry empty

closed 2016-05-15 by AndyR88

jjmccl, 2019-02-18:
I'm going to add this just because I spent about an hour trying to figure out how to log out/log in, restoring configs by hand etc to try to solve this same problem.  It's dumb of me, but I hope this helps someone else.  Just click the second icon in the top left (the that looks like a  little man).  Phew.