motioneye-project/motioneye #206: Motioneye crashes when connecting to rtsp with authentication on Phylink. Motioneyeos is fine.

brik2k, 2016-05-16:
Hi,
I'm running version 0.31.04 on a fresh install of Ubuntu Server 16.04 and I'm having an issue setting up motioneye with a Phylink network camera using rtsp and authentication. Every time I attempt to connect the camera I get an  "An error has occurred. Refresh recommended." If I restart motioneye it fails to start. Oddly, I've found that disabling authentication on the camera itself fixes the issue with no changes to the motioneye configuration files. I also have an Amcrest network camera that has no problem connecting with authentication. And, strangely, I have a rpi3 with motioneyeos that has no problem connecting to both cameras. 

I've turned logging up to All and reviewed the motion.log file. The only errors I see are 401 Not Authorized responses from the camera. I've verified the user/pw multiple times and have even used the same thread-1.conf file from the rpi. 

Any ideas?
Thanks

ccrisan, 2016-05-17:
Please attach your motioneye.log file. If you're under systemd, you'll get it using `journalctl -u motioneye > motioneye.log`. Otherwise you'll find it at `/var/log/motioneye.log`.

brik2k, 2016-05-17:
Here you go. I can also provide the camera login information if you'd like to test. Thanks
[motioneye.zip](https://github.com/ccrisan/motioneye/files/268825/motioneye.zip)

ccrisan, 2016-05-18:
It seems to be a problem with motion as it crashes soon after it starts. I recommend using [Mr. Dave's version of motion](https://github.com/Mr-Dave/motion).

Ethocreeper, 2016-05-25:
wait your cameras are rtsp any you can run them on raspberry pi 3b? if i tried to add a rtsp camera it says rtsp is not supported and @ccrisan said that motion daemon doesn't support it so he cant add support.
i found a modified version of motion demon [here](https://github.com/hyperbolic2346/motion)

ccrisan, 2016-05-25:
What did I say? Where did I say that?

Ethocreeper, 2016-05-25:
sorry am stubit it is rtmp which i requested [here](https://github.com/ccrisan/motioneye/issues/188) and that motion demon is for rtmp

now that i find one that support rtmp you will add suport???

ccrisan, 2016-05-25:
I can't add support for RTMP as it's something the motion daemon has to support and I'm not willing to dig into motion's source code.

closed 2016-07-03 by ccrisan