motioneye-project/motioneye #207: not able to add pi camera module v2.1 noir

guocity, 2016-05-21:
I have used the motioneyeos-raspberrypi3-20160410.img.gz image, with no problem,
however, i tried to follow the instruction to install motion eye in Raspbian, the camera module v2.1 noir can not be detected. 
Is the driver missing?

ccrisan, 2016-05-22:
Have you loaded the `bcm2835-v4l2` module?

guocity, 2016-05-26:
thanks ccrisan, it's the driver issue
is there any page shows how to setup IR for pi camera noir?

legotheboss, 2016-05-27:
Try running these commands... It fixed my issues with Raspberry Pi Zero 1.3 and Pi NoIR V2

`sudo apt-get update` 

`sudo apt-get dist-upgrade`

ccrisan, 2016-06-05:
Unless this is really about [motionEyeOS](https://github.com/ccrisan/motioneyeos), there shouldn't be a problem with any of the RPi cameras on an up-to-date Raspbian.

closed 2016-06-05 by ccrisan