motioneye-project/motioneye #208: Cameras do not fit vertically

jraydoran, 2016-05-23:
I have added 4 x raspberry pi cameras to a 5th raspberry pi 3 acting as a hub. I have the 4 x cameras in 2 columns. I cannot get the 4 camera's to fit on screen. I have to scroll down to see the bottom 2 cameras. I have selected the "Fit Frames Vertically" option on all the cameras and the hub.

Any advice?

Kind regards
Jason

ccrisan, 2016-05-24:
A screenshot would help. The Fit Frames Vertically option makes sure that at least one *row* of frames fits vertically on the screen.

jraydoran, 2016-05-25:
Hi ccrisan,

Thank you so much for the feedback. Attached is a screenshot.

Kind Regards
Jason

[image]

ccrisan, 2016-05-25:
Well yeap, as I said, the fit vertically option makes sure that one row of frames fits in the window boundaries. In your case I'd recommend setting 3 frames/row so that the frames are proportionally smaller and thus two rows will fit vertically in your window.

closed 2016-05-25 by ccrisan

jraydoran, 2016-05-25:
Thank you very much ccrisan.