motioneye-project/motioneye #211: resolution monitor and layout...

iSlide, 2016-05-24:
Hi,
I'm finishing to prepare my survaillance system.
I would like to connect to my raspberry pi3 a display for to check that all works fine.
I have a 15" display with resolution of 1024*768. Unfortunately I have little space and I can't use more big display.
With this resolutione MotionEye doesn't show the option of number of columns, but I've a layout in mobile style.
Can I force the system to works in standard mode with number of the columns in layout?
Many thanks to all for your help.
have a nice day

ccrisan, 2016-05-24:
You could use Ctrl & Minus to "zoom out" a bit.

iSlide, 2016-05-25:
Many thanks. With two times of Ctrl&Minus I resolved the problem. 
Many thanks again!

closed 2016-05-25 by iSlide