motioneye-project/motioneye #212: How to provide my external IP address on an alert from Motioneye

Hard2findaNameHere, 2016-05-27:
I have successfully installed motioneye on a raspberrypi3 and set up a static internal IP address.  Everything appears to be working great, remote access via the internet, alerts etc. 
I have also attempted to set up a NO-IP DDNS and associated it to my router and this works, but unfortunately I cannot install DUC (Dynamic Update Client) to advise NO-IP if my external IP address changes at anytime.

Is it possible that Motioneye could incorporate within its GUI and alerts and option to advise of the external IP address? Maybe only necessary if it changes but on all alerts it would be extremely useful.

bakroistvan, 2016-06-21:
I think this feature is not feasible to the motioneye.

Try https://duckdns.org (hackster friendly DDNS)
 - Sign in
 - Create subdomain
 - Click install on the top
 - Choose the domain
 - Click your OS eg.: pi
 - Follow instructions

ccrisan, 2016-07-03:
Such a feature has nothing to do with motionEye.

closed 2016-07-03 by ccrisan

Hard2findaNameHere, 2016-07-04:
Hi Calin,
When you eventually get around to implementing a few more action buttons could you let us know please. I really need them for the motioneye. Just simple buttons like UP, DOWN, LEFT and RIGHT would be fantastic and greatly appreciated.

----- Reply message -----

Closed #212.

---
https://github.com/ccrisan/motioneye/issues/212#event-711461458