motioneye-project/motioneye #213: netcam_read_rtsp_image: invalid frame!: Connection reset by peer

sebastientaggart, 2016-06-01:
Hello, I am using a HikVision DS-2CD3132F-IW camera (https://www.amazon.com/Hikvison-DS-2CD3132F-IW-4mm-Waterproof-Surveillance/dp/B019H2181G), and I've tied it into motionEye via RTSP, using this address: rtsp://192.168.84.201/Streaming/Channels/1

Everything works fine, including motion detection, for about 5 minutes, then I see this in the /var/log/motion.log:

```
[2] [ERR] [NET] netcam_read_rtsp_image: invalid frame!: Connection reset by peer
[2] [ERR] [NET] netcam_handler_loop: Error getting jpeg image
[2] [ERR] [NET] netcam_read_rtsp_image: invalid frame!: Connection reset by peer
[2] [ERR] [NET] netcam_handler_loop: Error getting jpeg image
[2] [ERR] [NET] netcam_read_rtsp_image: invalid frame!: Connection reset by peer
[2] [ERR] [NET] netcam_handler_loop: Error getting jpeg image
[2] [ERR] [NET] netcam_read_rtsp_image: invalid frame!: Connection reset by peer
[2] [ERR] [NET] netcam_handler_loop: Error getting jpeg image
[2] [ERR] [NET] netcam_read_rtsp_image: invalid frame!: Connection reset by peer
[2] [ERR] [NET] netcam_handler_loop: Error getting jpeg image
[2] [ERR] [NET] netcam_read_rtsp_image: invalid frame!: Connection reset by peer
[2] [ERR] [NET] netcam_handler_loop: Error getting jpeg image
[2] [ERR] [NET] netcam_read_rtsp_image: invalid frame!: Connection reset by peer
[2] [ERR] [NET] netcam_handler_loop: Error getting jpeg image
[2] [ERR] [NET] netcam_read_rtsp_image: invalid frame!: Connection reset by peer
[2] [ERR] [NET] netcam_handler_loop: Error getting jpeg image
[2] [ERR] [NET] netcam_read_rtsp_image: invalid frame!: Connection reset by peer
[2] [ERR] [NET] netcam_handler_loop: Error getting jpeg image
[2] [ERR] [NET] netcam_read_rtsp_image: invalid frame!: Connection reset by peer
[2] [ERR] [NET] netcam_handler_loop: Error getting jpeg image
[2] [ERR] [NET] netcam_read_rtsp_image: invalid frame!: Connection reset by peer
```

To "fix", I go into motionEye settings for this camera, disable Video Device, Apply, then enable, and it works again for another 5 minutes or so.

I'm running ``motion Version 3.2.12+git20140228`` and motionEye 0.31.4

Google wasn't very helpful in this case, has anyone run into this issue?

ccrisan, 2016-06-01:
I recommend using Mr Dave's motion version.

sebastientaggart, 2016-06-01:
Great, trying that now.

JHimmelbauer, 2016-06-02:
have currently the same problem with one rtsp camera (motionEye 0.31.5) - a restart of the motioneye service reconnect the camera again....

Mybe the problem is a short network connection lost - this cam is a WAN based IP cam, A other cam (same model) in the LAN don't has this problem

ccrisan, 2016-06-02:
Maybe this should be taken to the motion forums or to [Mr Dave's repo](https://github.com/Mr-Dave/motion/issues) as I wouldn't know what to do about it.

JHimmelbauer, 2016-06-02:
OK, thank you Sir- will check my luck with Mr.Dave ;-)

JHimmelbauer, 2016-06-07:
@sebastientaggart - do you have found a solution about this problem?

JHimmelbauer, 2016-06-07:
It looks like, if is this am old motion problem - see http://www.lavrsen.dk/foswiki/bin/view/Motion/BugReport2006x12x18x192713 - I have some cameras connected via WIFI - if the connection is stable I have no problem over days, but I can reproduce the problem if I reboot the WLAN bridge of the connected camera (http://www.verderber-retz.at) - think it is "only" a reconnect of motion problem. If I restart the motionEye service, the camera goes back online...

ccrisan, 2016-07-03:
I'm closing this issue since there's nothing motionEye can do about it.

closed 2016-07-03 by ccrisan