motioneye-project/motioneye #214: Adjusting brightness from shell script

stevegpage, 2016-06-01:
I am trying to flip the brightness and contrast levels automaticaly at sunset/sunrise, as auto brightness does not work on the Pi. My plan was to run 'sed' on the thread-1.conf and restart the motioneye service but this does not seem to reload the config file changes - is there another way to achieve this?

DawsonLeBrown, 2016-06-06:
Hi Steve,
I have a cron job which runs each hour throught the day and night to execute  shell script which does the following:

sudo perl -pi -e 's/auto_brightness off/auto_brightness on/g' /etc/motioneye/thread-1.conf 
sudo systemctl restart motioneye
sleep 30s
sudo perl -pi -e 's/auto_brightness on/auto_brightness off/g' /etc/motioneye/thread-1.conf
sudo systemctl restart motioneye

I'm using an rPi 3 with the noIR camera - this has so far proven to be enough to set the camera's exposure to a reasonable level each hour as the day's light conditions change. Rather than relying on dawn/dusk to change the settings, it also allows for changing meteorological conditions.
This works for me

stevegpage, 2016-06-06:
Thats a good idea - I did notice that after setting the auto brightness and restarting it reset things. I thought I had triggered motioned to read the config data - but could not repeat it, but now I think I can understand why. Will give this a go and see what happens this evening, thanks.

JamBalaya56562, 2026-11-09:
DawsonLeBrown's cron approach answered this, and there was no follow-up since 2016. On current versions brightness is a camera control in the UI, and the per-camera files are `camera-N.conf`. Closing; please open a new issue if you need this on a current version.

closed 2026-11-09 by JamBalaya56562